Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #4: Falsche Werte WW Temperatur

keldi23, 2024-01-14:
In Regelmäßigen Abständen werden unplausible Werte der Warmwassertemperatur aufgenommen.
Durch diese wird der Temperaturverlauf im Home Assistant Diagramm nicht richtig dargestellt.

[image]

Als Lösung kann der Code folgendermaßen abgeändert werden:

#Warmwasser-Temperaturabfrage 
    - can_id: 0x180
      then:
        - lambda: |-
            if(x[0]==0xd2 and x[1]==0x00 and x[3]==0x00 and x[4] == 0x0e) {
              float temperature =float((float((int((x[6])+( (x[5])<<8))))/10));
              if(temperature < 100) {
                id(temperature_water).publish_state(temperature);
                ESP_LOGD("main", "Temperature received over can is %f", temperature);
              }
            }

keldi23, 2024-01-14:
Das selbe bei TV und TVBH

Trunks1982, 2024-01-14:
Nein bei mir läuft das einwandfrei.Da stimmt bei dir was nicht.Ich habe in keinem Wert unplausible Werte.

keldi23, 2024-01-14:
Alles klar, ich weiß auch nicht warum. Vielleicht hängt das auch mit der WP Version zusammen oder so.
Ich lass die Änderung bei mir mal drin, dann siehts gut aus.

Auf jeden Fall vielen Dank für deine super Arbeit.

Trunks1982, 2024-01-14:
An der WP Version sollte es nicht liegen.Hast du denn jetzt die neuste yaml mal installiert?
Was benutzt du für ein Kabel zum ESP?
Verdrillt?

Danke für dein Feedback.

keldi23, 2024-01-14:
Ja, die neuste YAML hatte ich drauf.
Als Kabel habe ich nur ein Klingeldraht. Kann wirklich daran liegen. Da hast du recht.

Trunks1982, 2024-01-14:
Ich benutze ein gedrilltes Lan Kabel. Wäre aufjedenfall nen Versuch wert.

ap20132022, 2024-01-31:
bei mir kommt WW auch richtig an

closed 2024-02-08 by Trunks1982