Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #9: no connection with heat pump

tosomax, 2024-02-22:
Hi

I have just installed your software on a ESP32-S3-WROOM-1 bought from aliexpress.
The ESP32 is connected to a CAN transmitter bought from aliexpress (wcmcu-230).

I have a ehsxb08p50ba daikin/rotex heat pump.

I have followed your guide to connect everything together and to the J13 CAN port of the heat pump.

Everything up and running, but the ESP32 does not seem to receive any data from the CAN Bus

It keeps saying: 
send standard id=0x680 rtr=FALSE size 7
COmponent time took a long time for an operation (1.00 s)
Componets should block for at most 20-30ms

Do I have the wrong type of CAN transmitter? I cannot seem to find exactly the model you suggested.

Do you have any suggestion about other things I need to check?

Thanks

Trunks1982, 2024-02-22:
Hi ich benutze diesen https://amzn.eu/d/fRetaO8 Transciver.Ist aber so wie ich es sehe das gleiche.Welches Baujahr ist die Rotex und welche genau ist es?

tosomax, 2024-02-22:
The model of the heatpump Is Daikin EHSXB08P50PA. It should be a 2017/2018 model. I'm not really sure how I can diagnose the issue...maybe I've set up something wrong. Do you have any advice? For reference, I have already set up another esp32 with espaltherma and it is working, I'm definitely not an expert but I am not a total noob, so I believe that at least the connections are made correctly. I can think possibly of a detective CAN trasceiver or maybe this specific hpsu unit needs different CAN commands?

tosomax, 2024-02-22:
ESPAltherma doesn't do CAN, I know there is a fork that might be able to use CAN, but I haven't tried.

The ESP32 I'm using with espaltherma is a differnt one, I've bought specifically a S3 wroom for your software.
I've followed this guide https://esphome.io/guides/getting_started_command_line.html and then I have loaded your .yaml file

I can see the device from HomeAssistant, I can reach the web page, but I have no communication with the heatpump.
If you think it is worth to repeat the process I will do it

Trunks1982, 2024-02-22:
Ich würde noch mal über Homassist den esp neu erstellen und wenn der esp angezeigt wird die yaml per copy paste einfügen.
Wenn sie can hat muss das funktionieren.Letztes mal hatten wir auch einen User der zb falsch geflasht hat und danach hat es funktioniert.
Ist nur nen Anhaltspunkt.
Können Sie mir mal nen Foto von der Bedieneinheit von der Rotex posten?

Trunks1982, 2024-02-22:
[image]
so sollte das log dann aussehen wenn was empfangen wird

tosomax, 2024-02-22:
thanks I will try to reflash it and let you know :)

tosomax, 2024-02-22:
Reflashed through ESPHome: I have installed ESPhome in docker, then from the webgui I reflashed the basic esphome on ESP32, then I copypasted the yaml file and reflashed again.

I don't have that "time" error, but still it seems that I don't get any data from the CAN bus.
What I might try is to change cables, then I think i could buy another CAN transceiver, it might be defective, who knows.

Do you believe that my heat pump has a different protocol so the commands that are sent through the can bus are not recognized? Sorry I'm not an expert in this field, I hope I am clear

[image]

Trunks1982, 2024-02-22:
Was man machen könnte wäre das debuggen höher zusetzen und einmal zugucken wenn händisch was an der heizung verstellt wird ob daten reinkommen.

Trunks1982, 2024-02-22:
[image]
Hast du nach dem ersten flashen auf edit geklickt und dann alles aus meiner yaml kopiert und 
[image]
so eingefügt?

Trunks1982, 2024-02-22:
Und was du mal ausprobieren könntest verstell mal über die webbi oder über Homeassistant zb die Betriebsart oder Raumsoll und guck mal ob sich was an der Heizung ändert.

tosomax, 2024-02-22:
Yes I have copypasted as you described. Now I have added the option "verbose" to the log.
Tried to change the operating mode via webgui and this is the log: (no change on the heatpump unfortunately)
[image]
I have changed the water tank temperature on the heat pump manually, and nothing comes up on the log.

I might just try and buy another CAN transceiver from Amazon, in case this is broken.

Trunks1982, 2024-02-22:
Das ist echt merkwürdig.Würde vielleicht bevor du einen neuen transreceiver bestellst einmal nen anderes Kabel testen.Deine Heizung sollte aufjedenfall funktionieren.Meine sieht genauso aus. CAN L und H sind nicht vertauscht?
Nur um alles auszuschliessen.

tosomax, 2024-02-22:
Thanks for the help :)

Trunks1982, 2024-02-22:
> Thanks for the help :)

No Problem.

tosomax, 2024-02-22:
changed all cables, used wires from ethernet cable, no success. Even checked continuity with a multimeter.

Last question before buying another CAN transceiver, do I need to change some parameter on the HPSU to activate the CAn port J13 or should it always work?

Thanks again

Trunks1982, 2024-02-22:
Sollte immer funtkionieren.

tosomax, 2024-02-22:
I've googled a bit, it seems that theese small CAN transceiver are frequently knock offs and defective. I bought it from aliexpress, might be the case. Also, from what I've found I should be measuring about 2.5V between CAN-H or CAN-L and GND. I'm measuring 1.6V. I've just bought a Waveshare CAN transceiver from Amazon, hoping it will be legit (10 times the cost than aliexpress, but I don't want to wait another 2 weeks).

I'll let you know how it goes!

Trunks1982, 2024-02-22:
Ich habe auch den von waveshare und die sind top.Bei amazon kann man den wenigsten schnell zurück schicken.👍

tosomax, 2024-02-25:
The waveshare can trasceiver arrived. Now I measure 2.5v between GND an CANH or CANL, so I Guess the other Can transceiver was a knock off or broken.  However, I cannot still get it working, I dont get any signal from the HPSU. 
I dont know if you might any advice...

I was wondering if I Need to connect the ESP32 to the ground of the J13 port or not? I am powering the ESP32 with an external Power supply

Trunks1982, 2024-02-25:
> Der Waveshare-Kann-Transceiver ist angekommen. Jetzt messe ich 2,5 V zwischen GND und CANH oder CANL, also denke ich, dass der andere Can-Transceiver ein Knock-off oder kaputt war. Ich kann es jedoch immer noch nicht zum Laufen bringen, ich bekomme kein Signal von der HPSU.Ich weiß nicht, ob du einen Rat geben könntest...
> 
> Ich habe mich gefragt, ob ich den ESP32 mit dem J13-Anschluss verbinden muss oder nicht? Ich steige den ESP32 mit einem externen Netzteil

Der Esp32 muss mit dem Waveshare verbunden werden und dann von da zur Rotex.War schon so richtig wie du es angeschlossen hattest.Hast du die richtigen Pins verwendet die ich in der yaml eingestellt habe?

tosomax, 2024-02-25:
Yes I used pins 47 and 48, I used also the same color code for the cables to avoid any mistake. I Will try ti repeat the flashing and double check everything once again just to be sure!

Trunks1982, 2024-02-25:
Werte verändern funktioniert auch nicht?
Hätte jetzt echt gedacht das es am transceiver liegt.Mhm

Trunks1982, 2024-02-25:
Ich hätte noch ne idee.

tosomax, 2024-02-25:
No unfortunately even changing values doesn't do anything.

I also thought It was the transceiver, but I Will do more experiments until I can get to the bottom of this :)

Trunks1982, 2024-02-25:
Wenn du den neu flasht steht das gleiche wie hier vei die in der grund yaml?

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

tosomax, 2024-02-25:
[image]

Trunks1982, 2024-02-25:
Meine wenn du den neu über esphome flasht ohne meine yaml vorher eingespielt zu haben?

tosomax, 2024-02-26:
Yes I have checked by reflashing, everything is right.

This Is out of my league, but could It be possibile that my heat pump used different "strings" and different "can id"? Is it possibile to write an easy code to have the ESP32 in listening mode and listen to what the can bus of the HPSU Is sending?
I don't know if you are familiar with PyHPSU, It Is something similar to your program with Raspberry Pi. I tried to check all the "can codes" and they look similar but not the same to yours. Maybe my heat pump is just different than yours

Trunks1982, 2024-02-26:
Wenn du debug auf verbose änderst und an der wärmepumpe einen wert änderst sollten werte reinkommen.

tosomax, 2024-02-26:
I have managed to get it working!! Thanks so much!

In the end I had a bad transceiver (solved with the new Waveshare) and with a multimeter I was able to find a faulty cable!!

Now I only need to translate your code to italian or english :D

I hope I can do it with google translate, change the yaml and have my own local version ;)

Trunks1982, 2024-02-26:
Das freut mich das es hetzt funktioniert.Hätte nicht gedacht das es an einem kabel liegt.

Trunks1982, 2024-02-26:
Wenn du eine Italienische bzw Englische Version hast kann ich die hier ja gerne veröffendlichen.

Trunks1982, 2024-02-27:
Danke für die Wertschätzung.
Das war auch mein gedanke so einfach wie möglich.PyHPSU und espaltherma haben mir nie zugesagt und deswegen bin ich auf diese Lösung gekommen.

mschallar, 2024-03-05:
I have the same issue.

I ordered the same waveshare as you did: https://www.amazon.de/dp/B00KM6XMXO and the the recommended ESP32 https://www.amazon.de/dp/B0CLD4QKT1.

The cables looked good with my multimeter. I checked the contunuity between the ESP32 and waveshare, as well as between the waveshare and J13 connector.
I also replaced each cable just to be sure, and checked the contunuity again. It did not work.

```
[23:09:36][W][component:214]: Component time took a long time for an operation (1.00 s).
[23:09:36][W][component:215]: Components should block for at most 20-30ms.
[23:09:36][D][sensor:094]: 'Thermische Leistung': Sending state nan kW with 2 decimals of accuracy
[23:09:36][D][canbus:035]: send standard id=0x680 rtr=FALSE size=7
```

I also have the Rotex RoCon B1 installed in my living room. That device still receives the correct values from the HPSU in the basement.

How can I debug this any further? Has anyone an idea?

tosomax, 2024-03-05:
You can test your transceiver by measuring the voltage between CANH and CANL, you Need to see some voltage when the ESP32 is trasmitting (once you boot up the esp32 with the code, It should start to transmit in a few seconds)

Then you can check if the transceiver is legit by measuring the voltage between CANH and GND, and also CANL GND. It should be around 2.5V when the Esp32 Is not trasmitting (to do so, disconnect CRX and TRX while leaving connected GND and 3.3V)

I can also suggest checking the cables between CANH and CANL to the heat pump: the connections should be CANH to CANH and CANL to CANL. The picture posted in GitHub can be misleading for someone who doesn't know the CAN protocol.

Let me know

mschallar, 2024-03-07:
Thanks @tosomax . I soldered all cables to the ESP32 to make sure they are properly connected. After I've done that, it worked right away.
Thanks @Trunks1982 for the effort!

closed 2024-04-07 by tosomax