Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #12: WW Soll je Zeitprogrammm

RaZER898, 2024-03-26:
Hallo, ich nutze WaveShare Esp32-S3 sowie Waveshare SN65HVD230 CAN Board an einer daikin altherma 3 R ECH2O (EHSX08P50DA3).

Die Einstellung für Warmwasser Soll bezieht sich ja auf die Temperatur des Warmwassers im Zeitraum 1 der Warmwasser automatik 1. Gibt es eine Möglichkeit, dass aktuelle Zeitprogramm bzw. die Zeiträume 1-3 separat anzusteuern.

Danke

Trunks1982, 2024-03-26:
> Hallo, ich nutze WaveShare Esp32-S3 sowie Waveshare SN65HVD230 CAN Board an einer daikin altherma 3 R ECH2O (EHSX08P50DA3).
>
> Die Einstellung für Warmwasser Soll bezieht sich ja auf die Temperatur [...]

Hi die möglichkeit bestünde.Müsste nur soll2 und 3 implementieren.

RaZER898, 2024-03-28:
Ich habe mir anderweitig helfen können, indem ich die Zeitprogramme an der Daikin so umgestellt habe, dass  das Zeitprogramm 1 immer die Zeiten betrifft wo ich die Temperatur variieren will. In meinem Fall in der Zeit wo viel PV Strom vorhanden ist. Danke dennoch für deine Mühe und Antwort

closed 2024-03-28 by Trunks1982