Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #30: "Thermische Leistung" incorrect for DHW

svde, 2024-04-08:
The "Thermische Leistung" performance numbers don't seem to be correct for the DHW program. The values are negative, while the heatpump is working at full power. See graph:

[image]

For heating it seems to be accurate: 
[image]

Trunks1982, 2024-04-08:
[image]

Also bei mir sieht es gut aus.Hatte ichvaichvnochvnie anders.Nur kurz wenn er anfängt.

Trunks1982, 2024-04-08:
My Rotex HPSU Compact mit 8kw aussen

svde, 2024-04-08:
So the same generation model. Interesting...

Nitrox912, 2024-04-08:
Das Problem habe ich auch bei meiner Daikin Altherma 3 HHT 14kw. Das kommt m. E. immer dann, wenn die WP anfängt zu laufen und dann mitunter das Heizungswasser im Rücklauf kurzzeitig wärmer ist, als im Vorlauf. Ich habe mir vorerst damit geholfen, erst bei einer Leistungsaufnahme von mehr als 350W die thermische Energie berechnen zu lassen. So hat die Pumpe erst einmal Zeit das Wasser umzuwälzen. Die Berechnung erfolgt dann somit erst, wenn der Kompressor auch richtig läuft. Erste Tests sahen ganz gut aus.

[image]

svde, 2024-04-09:
[image]

Trunks1982, 2024-04-09:
Meiner Meinung nach darf tvbh nicht niedriger sein als TR wärend er WW macht.

Trunks1982, 2024-04-09:
> Das Problem habe ich auch bei meiner Daikin Altherma 3 HHT 14kw. Das kommt m. E. immer dann, wenn die WP anfängt zu laufen und dann mitunter das Heizungswasser im Rücklauf kurzzeitig wärmer ist, als im Vorlauf. Ich habe mir vorerst damit geholfen, erst bei einer Leistungsaufnahme von mehr als 350W die thermische Energie berechnen zu lassen. So hat die Pumpe erst einmal Zeit das Wasser umzuwälzen. Die Berechnung erfolgt dann somit erst, wenn der Kompressor auch richtig läuft. Erste Tests sahen ganz gut aus. > > [image] Ist das Heizen oder WW?
Wenn das WW ist stimmt da was nicht.Zu wenig Leistung.

Nitrox912, 2024-04-09:
@svde Meines Erachtens berücksichtigt die Formel für die Berechnung der Heizleistung die falsche Vorlauftemperatur. TVBH ist der Heizungsvorlauf. Ich will ja aber die Heizleistung der WP errechnen. Deshalb habe ich den Vorlauf (TV1) direkt vor dem Wärmetauscher und hinter dem Wärmetauscher (TR2) meiner Berechnung zugrunde gelegt. Deshalb haut es m. E. bei dir beim Heizen hin, aber nicht beim WW.

Nitrox912, 2024-04-09:
> > Das Problem habe ich auch bei meiner Daikin Altherma 3 HHT 14kw. Das kommt m. E. immer dann, wenn die WP anfängt zu laufen und dann mitunter das Heizungswasser im Rücklauf kurzzeitig wärmer ist, als im Vorlauf. Ich habe mir vorerst damit geholfen, erst bei einer Leistungsaufnahme von mehr als 350W die thermische Energie berechnen zu lassen. So hat die Pumpe erst einmal Zeit das Wasser umzuwälzen. Die Berechnung erfolgt dann somit erst, wenn der Kompressor auch richtig läuft. Erste Tests sahen ganz gut aus. > > [image] Ist das Heizen oder WW?
> > Wenn das WW ist stimmt da was nicht.Zu wenig Leistung.
> > Es sei denn WW ist schon bei 50Grad dann ist klar dann hat man weniger Thermische Leistung.

Das war nur zu Testzwecken und die WW-Temperatur war schon bei knapp unter 50 Grad.

Nitrox912, 2024-04-09:
```
template:
  - sensor:
      - name: "Heizleistung Wärmepumpe"
        unique_id: wp_heizleistung
        unit_of_measurement: "W"
        state: >
          {% set kompressor_leistung = states('sensor.shellypro3em_3ce90e6dd41c_total_active_power') | float %}
          {% if kompressor_leistung >= 350 %}
            {{ ((states('sensor.rotex_heizkreis_vorlauf_tv') | float - 
                 states('sensor.rotex_ruecklauftemperatur_heizung') | float) * 
                 states('sensor.rotex_durchfluss') | float * 
                 4.186 / 3600 * 1000) | round(2) }}
          {% else %}
            0
          {% endif %}
```

Trunks1982, 2024-04-09:
Tv1 und tr1 ist defintiv falsch.😉

Nitrox912, 2024-04-09:
> Tv1 und tr1 ist defintiv falsch.😉

Weil?

Trunks1982, 2024-04-09:
Sogar nen Rotex Mitarbeiter bzw Service Mensch so mit meiner Formel rechnet.Und in den Foren stehts auch so.Wo du recht hattest war 4,91 falsch.Hatte ich wohl falsch übernommen.

Nitrox912, 2024-04-09:
Also, wenn ich richtig aufgepasst habe, dann benötige ich zur Berechnung der thermischen Heizleistung 3 Werte:

1.  Volumenstrom der Heizung - den Wert haben wir
2. Temperaturdifferenz zwischen Vor- und Rücklauf - diese Ermitteln wir durch VT-RT
3.  Spezifische Wärmekapazität von Wasser = 4,186 

TVBH ist aber Vorlauf nach Heizungswärmetauscher. Der spielt meiner Auffindung nach aber für die direkte Berechnug der Heizleistung der WP aber keine Rolle. Entscheidend hierfür ist die Temperaturdifferenz des Wassers direkt vor und nach der Wärmepumpe.

Trunks1982, 2024-04-09:
Also Berechnest du ohne Verluste und dadurch wirds schöner.Und wo nach regelt die interne Umwälzpumpe?
Nicht nach TV sondern nach TVBH.Aber nun gut muss jeder für sich selber entscheiden.Ich will es nicht geschönt haben.
Für mich ist Leistung nach Wärmetauscher ausschlaggebent.

Trunks1982, 2024-04-09:
TVBH darf wärend des Vorgangs nicht unter TR sein.Dann ist was nicht an der Heizung ok.Könnt ihr aber mal im Heizungsforum erfragen.Zum Ende der WW Bereitung sind die minimal bei einander. Würde sagen ca 1-2Grad sonst nicht.

Nitrox912, 2024-04-09:
> Also Berechnest du ohne Verluste und dadurch wirds schöner.Und wo nach regelt die interne Umwälzpumpe? Nicht nach TV sondern nach TVBH.Aber nun gut muss jeder für sich selber entscheiden.Ich will es nicht geschönt haben. Für mich ist Leistung nach Wärmetauscher ausschlaggebent.

Die Regelung der Umwälzpumpe spielt ja insofern nur eine Rolle um zu berechnen, wie viel Wärme gerade durch die Leitungen transportiert wird. 

Bei der Produktion von Heizungswärme stimme ich dir uneingeschränkt zu, aber sind denn deine Werte dann bei der Warmwasserproduktion logisch oder hast du da ähnliche Abweichungen? 

Ich erstelle mir mal ein zweites Template mit TVBH und vergleiche die Unterschiede beim Heizen und in der WW-Bereitung.

Trunks1982, 2024-04-09:
Welche Abweichungen meinst du?
Also bei WW Bereitung hab ich immer Abstand von TVBH zu TR.Beim Anfang nicht ist klar.In der Regel hab ich nen Abstand von 5K.Das das bei höheren WW Temps nicht möglich ist, ist auch klar.Sollte dennoch abstand haben.

Nitrox912, 2024-04-09:
> Welche Abweichungen meinst du? Also bei WW Bereitung hab ich immer Abstand von TVBH zu TR.Beim Anfang nicht ist klar.In der Regel hab ich nen Abstand von 5K.Das das bei höheren WW Temps nicht möglich ist, ist auch klar.Sollte dennoch abstand haben.

Die Abweichungen weswegen @svde dieses Issue aufgemacht hat.

Trunks1982, 2024-04-09:
Nein siehst ja auf meinem Bild oben.Das war WW Bereitung bei 48 Grad WW Temp

Trunks1982, 2024-04-09:
Guckt mal bitte nach was ihr für eine Hysterese bei Umwälzpumpe eingestellt habt.kann man glaub nur im Service Menü nachgucken.5k sollte es sein.

Nitrox912, 2024-04-09:
Du meinst die Differenz zwischen Vor- und Rücklauf? Bei mir nennt sich das Pumpe dT Heizen =5K

Trunks1982, 2024-04-09:
> Du meinst die Differenz zwischen Vor- und Rücklauf? Bei mir nennt sich das Pumpe dT Heizen =5K

Ja genau das gleiche gibts auch für ww im service menü.

Trunks1982, 2024-04-09:
Wie gesagt das für die Pumpe gibt es einmal für Heizen und für WW.

Nitrox912, 2024-04-09:
> Wie gesagt das für die Pumpe gibt es einmal für Heizen und für WW.

Hab noch mal geguckt. Bei Daikin gibt es das tatsächlich nicht. Da gilt der Wert für beides.

Trunks1982, 2024-04-09:
Ah ok.Bei mir und bei der neueren Rotex Ultra gibts das im Service Menü.Nicht Fachmann Menü

Nitrox912, 2024-04-09:
Also: Heizung spielt bei dem kurzen Test keine Rolle, welche VL-Temperatur man nutzt. Die sind nahezu gleich.

Aber: Wenn man sich mein System angucktn, wird TVBH wird bei der Warmwasserbereitung zur Rücklauftemperatur, weil sie nach dem Speicher gemessen wird. So können natürlich keine vernünftigen Werte errechnet werden und es passiert das gleiche wie bei @svde 

Die Delle bei WW einfach ausblenden. Da ist der ESP abgestürzt und hat die WW-Bereitung unterbrochen.

[image]

[image]

Trunks1982, 2024-04-09:
Hast ja recht anfangs ist das auch so das tvbh zur tr wird.Nach 15-20min sollte das aber nicht mehr sein.
Und svde hat die gleiche Heizung wie ich also müsste das bei ihm gehn.

Nitrox912, 2024-04-09:
> So hab in die yaml einmal die Abfrage Spreizung mod HZ und WW eingebaut.Bitte einmal testen ob ihr die auch reinbekommt.

Werte stehen bei mir drin und betragen jeweils 7K. Kann dies aber in keiner Einstellung finden und werden m. E. auch nicht berücksichtig. Die Pumpe versucht bei mir immer die eingestellten 5 K unter Pumpe dT zu erreichen.

Trunks1982, 2024-04-09:
Ah ok.Dann ist es für deine Anlage nicht.Ich glaube wir sollten mal aufschreiben was bei deiner Anlage geht und nicht geht.Dann könnte man Vielleicht deine und unsere deferenzieren.

Nitrox912, 2024-04-09:
Gibt es die Möglichkeit das auch nicht über GitHub zu diskutieren, sondern einem richtigen Chat?

Trunks1982, 2024-04-09:
Mir würde jetzt nur Telegram einfallen.

svde, 2024-04-09:
Oder Signal?

Nitrox912, 2024-04-09:
Mir egal, irgendeine Gruppe wäre gut.

Trunks1982, 2024-04-09:
Das hab ich nicht.

https://t.me/+C1iVhAjaENw0ODY0
Hab da mal ne gruppe erstellt

svde, 2024-04-09:
[image]

svde, 2024-04-09:
- platform: template
    name: "Thermische Leistung"
    device_class: "power"
    unit_of_measurement: "kW"
    accuracy_decimals: 2
    update_interval: 10s
    lambda: |-
      if (id(status_kessel).state == 1) {
          const float TVBH_state = id(TVBH).state;
          const float ruecklauf_state = id(ruecklauf).state;
          const float durchfluss_state = id(durchfluss).state;

          return ((TVBH_state - ruecklauf_state) * (4.91 * durchfluss_state)) / 3600;
      } else {
          return 0.0; // oder einen anderen Wert, wenn der Kompressor nicht aktiv ist
      }

Nitrox912, 2024-04-09:
Probiere das mal und kopiere den Code in deine configuration.yaml von HA

```
template:
  - sensor:
      - name: "Heizleistung Wärmepumpe"
        unique_id: wp_heizleistung
        unit_of_measurement: "W"
        state: >
          {{ ((states('sensor.rotex_heizkreis_vorlauf_tv') | float - 
               states('sensor.rotex_ruecklauftemperatur_heizung') | float) * 
               states('sensor.rotex_durchfluss') | float * 
               4.186 / 3600 * 1000) | round(2) }}
```

Trunks1982, 2024-04-09:
> ```
>   - platform: template
>     name: "Thermische Leistung"
>     device_class: "power"
>     unit_of_measurement: "kW"
>     accuracy_decimals: 2
>     update_interval: 10s
>     lambda: |-
>       if (id(status_kessel).state == 1) {
>           const float TVBH_state = id(TVBH).state;
>           const float ruecklauf_state = id(ruecklauf).state;
>           const float durchfluss_state = id(durchfluss).state;
> 
>           return ((TVBH_state - ruecklauf_state) * (4.91 * durchfluss_state)) / 3600;
>       } else {
>           return 0.0; // oder einen anderen Wert, wenn der Kompressor nicht aktiv ist
>       }
> ```
> 
> Shouldn't this be TV - TR instead of TVBH - TR?

Nein.Bzw kannst machen verschönigt nur das ganze ohne Verluste.

closed 2024-05-07 by Trunks1982