Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #45: Encryption Key

adieball, 2024-07-01:
Hi

when I try to add the flashed ESP32 (auto detected in HA), HA is asking me for an API Encryption Key.
As far as I can see nothing has been specified.

Also: anything else I need to do? I see the yamlfile for HA, so I need to import?
Is there a HA Forum thread, couldn't find one

thanks

Trunks1982, 2024-07-01:
Hallo, hast du esphome in HA installiert?
Wenn ja kannst du über file editor in esphome/secret den api key selber eintragen. Ich muss das noch in die Doku mit aufnehmen.

Wenn du das gemacht hast sollte sich der esp automatisch in HA anmelden.
Sonst brauchst nichts weiter tun.
Nein es gibt keinen HA Thread. Hier gibt es ja die möglichkeit.

adieball, 2024-07-01:
Hi. Danke für das schnelle Feedback.
Ja, ESPHome ist installiert, der ESP32 wird auch sofort zur Config angezeigt, will aber den Key haben.,

Interessanterweise habe ich noch nichtmal ein esphome directory

Trunks1982, 2024-07-01:
Ja pass auf erstell mal ein gerät ohne es zu flashen.Dann bekommst auch nen key.

adieball, 2024-07-01:
Ich bin ein Idiot, habe übersehen, dass das ESPHome Dashboard ein "AddOn" ist .....
installiere gereade

Trunks1982, 2024-07-01:
Alles gut🤣

adieball, 2024-07-01:
I'm lost ... Ich glaube ich verstehe das Prinzip von ESPHome nicht.

ESPHome Addon installiert.
Dann auf "New Device", wo es mich zu meinem WLAN abfragt und danach eine Menge Zeug lädt und kompiliert.

Aber ich habe die ROTEX Software doch schon auf dem ESP drauf.

HA findet den ESP "Rotex" ja auch, will aber einen Key

Trunks1982, 2024-07-01:
Brauchst nicht compilieren lassen.Macht einfach skip und er zeigt dir nen key an.

adieball, 2024-07-01:
Auch schon probiert. Wenn ich den dann nutzen will, um den von HA gefundenen ESP32 hinzuzufügen, sagt HA mir "falscher Key".

Würde für eine Remote Session auch einen "Kaffee" springen lassen :-)

Trunks1982, 2024-07-01:
Doch den kannst nutzen.musst gerät löschen. Und den key unter file editor esphome/secret.yaml eintragen.

Kann leider nicht bin arbeiten.

Trunks1982, 2024-07-01:
api_encryption_key: “hier key eintragen“

das muss rein.

adieball, 2024-07-01:
Habe jetzt die yaml datei von guthub genommen und in die erstellte config kopiert und dann wirelessly übertragen .... aus welchem Grund auch immer funktioniert es nun :-)

nun mache ich mal mit dem CAN Modul weiter und versuche es an die Heizung zu bauen .... 

Danke für die schnelle Hilfe!

Trunks1982, 2024-07-01:
Die ist eigentlich die selbe datei.Hat aber bei dir dann glaub den key erstellt.