Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #48: Installationsprobleme bei Schritt 7

stasihasi, 2024-07-15:
Hallo zusammen,

bin gerade dabei den ESP32 zu installieren. Die Schritte 1 bis 6 haben soweit auch gut funktioniert. Auch das Wlan Rotex Fallback Hotspot wird angezeigt und ich kann mich auch mit dem Passwort verbinden.

Jetzt zum Problem:
Leider öffnet sich aber nicht automatisch ein Browserfenster auf der ich mein WLAN eintragen kann.

Hoffe ihr könnt mir weiterhelfen!

Viele Grüße!

MichaelGer1410, 2024-07-15:
Wenn du mit dem Hotspot verbunden bist, einfach mal http://192.168.4.1 in den brwoser eingeben.

stasihasi, 2024-07-15:
ich bin geflasht von der schnellen Rückmeldung! 
Funktioniert! Am besten wird die Anleitung um die IP-Adresse ergänzt ;-)
Bin hoffentlich nicht der einzige, bei dem das passiert.

Vielen Dank nochmal!

MichaelGer1410, 2024-07-15:
Hatte das Problem anfangs auch, daher der Tipp ;)
Schön das es bei dir Funktioniert!

stasihasi, 2024-07-18:
Problem gelöst:
http://192.168.4.1/

closed 2024-07-18 by stasihasi