Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #49: Kühlbetrieb: keine Werte bei Umwaelzpumpe, Status Kompressor und Thermische Leistung

stasihasi, 2024-07-20:
Hallo zusammen,

der ESP32 läuft soweit ganz gut seit ein paar Tagen. Was mir jedoch aufgefallen ist, dass beim Kühlbetrieb (andere Modi hab ich aufgrund der Jahreszeit noch nicht benutzt) für:
- Umwaelzpumpe (welche Pumpe ist gemeint, die für Trinkwasserzirkulation? Diese ist bei uns extern verbaut)
- Status Kompressor
- Thermische Leistung (liegt es hier am Kühlmodus selbst?)

keine Werte abgerufen werden.

Anbei der entsprechende Screenshoot!
[image]

Trunks1982, 2024-07-20:
Umwälzpumpe ist die intern verbaute.
Thermische Leistung beim Kühlen ist auf 0 gesetzt. Normal liegt der wert im Minus Bereich.
Status Kompressor ist beim Kühlen definitiv an. Ich hab aber eine Rotex HPSU Compact, da wird es unterstützt.

stasihasi, 2024-07-20:
Die Bezeichnung des Innengeräts ist bei mir: Altherma 3 R ECH2O (EHSX08P50DA3)

Bzgl:
- Umwaelzpumpe: OK, also wie schon vermutet
- Status Kompressor: Schade dann, dass es bei meinem Modell anscheinend nicht funktioniert
- Thermische Leistung: also möchtest du nicht, dass der Wert negativ angezeigt wird und setzt ihn auf 0 wenn er negativ wird? Wäre halt ganz interessant für den Kühlbetrieb auch diese Info zu haben, aber ist auch nicht so tragisch.

Trunks1982, 2024-07-20:
Kannst du gerne bei dir im code ändern wenn du möchtest.Vielleicht änder ich das auch wieder.Mal schaun

stasihasi, 2024-07-20:
kannst du kurz beschreiben, wie ich das machen kann?

Trunks1982, 2024-07-20:
der code müsste angepasst werden : - platform: template
    name: "Thermische Leistung"
    device_class: "power"
    unit_of_measurement: "kW"
    accuracy_decimals: 2
    update_interval: 10s
    lambda: |-
      if (id(text_betriebsart).state == "Warmwasserbereitung") {
        return ((id(TV).state - id(ruecklauf).state) * (4.19 * id(durchfluss).state)) / 3600;
      } else if (id(text_betriebsart).state == "Heizen") {
        return ((id(TVBH).state - id(ruecklauf).state) * (4.19 * id(durchfluss).state)) / 3600;
      } else {
        return 0.0;
      }

Trunks1982, 2024-07-20:
> Die Bezeichnung des Innengeräts ist bei mir: Altherma 3 R ECH2O (EHSX08P50DA3)
>
> Bzgl:
>
> * Umwaelzpumpe: OK, also wie schon vermutet
> * Status Kompressor: Schade dann, dass es bei meinem Modell [...]

Danke für deine Spende!👍

stasihasi, 2024-07-20:
> der code müsste angepasst werden : - platform: template name: "Thermische Leistung" device_class: "power" unit_of_measurement: "kW" accuracy_decimals: 2 update_interval: 10s lambda: |- if [...]

Wo trage ich den Code ein, einfach in der configuration.yaml oder an einer anderen Stelle?

Trunks1982, 2024-07-20:
Über das addon esphome.Da müsstest aber alles sehn vom code.

stasihasi, 2024-07-22:
hab das ESPHome Add-on erst jetzt installiert. Der ESPHome war vorher und ist aktuell noch über die Integration mit HomeAssistant verbunden. Jetzt weiß ich aber nicht, wie ich einen fertig eingerichteten ESP nachträglich in das Add-on ESPHome bekomme, um mir dort dann den Code zu bearbeiten:
[image]

Trunks1982, 2024-07-22:
Erstellst einen neuen esp drin und kopierst vom github die yaml in den editor.Was vorher dein steht bitte rauslöschen.

stasihasi, 2024-07-22:
ah ok, zerschieß ich mir dann das bisherige? Hab schon Automationen erstellt und würde ungern die ganze Arbeit nochmal machen wollen

stasihasi, 2024-07-22:
ok danke!

stasihasi, 2024-07-22:
So würde ich dann den Code anpassen, um die thermische Leistung im Kühlmodus zu erhalten:
- platform: template
    name: "Thermische Leistung"
    device_class: "power"
    unit_of_measurement: "kW"
    accuracy_decimals: 2
    update_interval: 10s
    lambda: |-
      if (id(text_betriebsart).state == "Warmwasserbereitung") {
        return ((id(TV).state - id(ruecklauf).state) * (4.19 * id(durchfluss).state)) / 3600;
      } else if (id(text_betriebsart).state == "Heizen") {
        return ((id(TVBH).state - id(ruecklauf).state) * (4.19 * id(durchfluss).state)) / 3600;
      } else {
        **return ((id(TVBH).state - id(ruecklauf).state) * (4.19 * id(durchfluss).state)) / 3600;
        #return 0.0;**
      }

Um die Anpassung wirksam zu machen, reicht nicht speichern zu klicken, sondern zusätzlich install:
Muss hier noch was beachtet werden, bevor install gekllickt wird, oder geht das einfach im laufenden Betrieb?
[image]

Trunks1982, 2024-07-22:
Dein code ist falsch.Guck mal hin

stasihasi, 2024-07-22:
also ich will haben, dass die Leistung negativ ist!
dachte ich muss die gleiche formel wie beim heizen nehmen und dann wirds beim Kühlen einfach negativ.

stasihasi, 2024-07-22:
[image]

geht 2x "else if" hintereinander?

Trunks1982, 2024-07-22:
Joa.Siehst ja auch weiter oben im code zb.
Sieht schon besser aus.Kannst ja so mal testen.

Trunks1982, 2024-07-23:
Und hats geklappt?

stasihasi, 2024-07-23:
mit der Hilfe von dir natürlich ;-)

[image]

Danke nochmal!

Trunks1982, 2024-07-23:
👍

closed 2024-07-23 by Trunks1982