Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #57: Anzeige Wärmemenge

Steinmetz1957, 2024-11-01:
Hallo, bei meiner Anlage passen die über CAN Bus ausgelesenen Werte für die erzeugte Wärmemenge Heizung und die erzeugte Gesamtwärmemenge nicht mit den am Display angezeigten Werten zusammen. Die über den Bus gelesenen Werte sind viel zu klein.
Kann es sein das eine falsche Adresse ausgelesen wird?
[image]

Steinmetz1957, 2024-11-01:
Hi, die Anlage hat die Bezeichnung EHSX08P50DA3. Der Wert für die erzeugte Energie WW passt genau zur Statistik den die Anlage im Display mir zeigt. Die Anlage läuft jetzt 3 Jahre und einen Monat. Die Stastistik zeigt mir 37000 KWh bzw 47000 KWh für Heiz- bzw. Gesamtenergie. Diese Werte sind auch plausibel. Ich stehe da im Moment vor einem Rätsel warum die Gerätestaistik etwas ganz anderes anzeigt wie der CAN Bus.

Steinmetz1957, 2024-11-01:
Habe wohl gerade einen falschen Button gedrückt aus versehen.

Trunks1982, 2024-11-01:
Ah ist ne Altherma. Mir ist nicht bekannt das die andere Werte anzeigt. Wenn du lust hast kannst gerne mal in den Chat reinkommen. Gut möglich das da noch einer mit einer Altherma ist der das Problem hat.

Steinmetz1957, 2024-11-01:
Wo gibt es eigentlich eine offizielle Quelle für die Hex Adressen und deren Bedeutung oder sind diese ausschließlich rückwärts durch auslesen ermittelt?

Trunks1982, 2024-11-01:
Es gibt in dem Sinne keine Liste.Es hat mal angefangen durch rücklesen der Adressen.

Steinmetz1957, 2024-11-01:
OK das habe ich vermutet. Leider ist Daikin - Rotex nicht so auskunkftsfreudig wie andere Hersteller und der technische Support ist in solchen Fragen nach meiner Erfahrung unterirdisch. Trotzdem eine gute und effiziente Hardware.
Danke für die prompten Antworten. Ich schau mal ob ich das selbst lösen kann, wenn nicht werde ich es noch einmal im Chat versuchen.

Trunks1982, 2024-11-01:
Über das log solltest du eigentlich sehn wenn du auf den Menü Punkt gehst was für ne Adresse rauskommt. Die muss aber dann in den Code implementiert werden.Geht nicht so über die yaml.

Steinmetz1957, 2024-11-01:
das wäre der Abschnitt der Log Datei der die falschen Zahlen zeigt.
[logs_daikin-rotex_logs(1).txt](https://github.com/user-attachments/files/17599694/logs_daikin-rotex_logs.1.txt)

Steinmetz1957, 2024-11-01:
ja, wo und wie kann ich eine andere Log erzeugen?

wrfz, 2024-11-01:
```
[13:43:25][I][sendGet:169]: EHS fuer CH can_id<0x680> command<31 00 FA 09 20 00 00>
[13:43:25][I][handle :169]: EHS fuer CH<45.000000> can_id<0x180> data<D2 00 FA 09 20 00 2D>
```
Die erste Zeile ist ein Request an die Pumpe.
Die drei Bytes `0xFA 0x09 0x20` sagen der Pumpe, dass sie `EHS fuer CH` zurück schicken soll.

Die zweite Zeile ist eine Antwort dazu. `handle` bedeutet, dass ESP die Antwort erfolgreich verarbeiten konnte.
Bei dieser Entity sind die letzten zwei Bytes der Antwort der Wert, den die Pumpe liefert.
`0x00 0x2D` entspricht dem dezimalen Wert `45`.

Ich sehe aktuell zwei mögliche Gründe dafür:
1) Dein Altherma Gerät versteht unter `0xFA 0x09 0x20` etwas anderes als `EHS fuer CH`.
2) Ist einer Fehler in der Pumpe. Eventuell ein Fehler in der Firmware.

wrfz, 2024-11-01:
> Wo gibt es eigentlich eine offizielle Quelle für die Hex Adressen und deren Bedeutung oder sind diese ausschließlich rückwärts durch auslesen ermittelt?

Wie @Trunks1982 geschrieben hat, gibt es keine offezielle Protokoll-Dokumention.
Es gibt allerdings zwei Projekte, die seit längerem existieren und von Nutzern verschiedenster Rotex/Daikin Modelle (mit CAN) eingesetzt wird (unteranderem auch Altherma Geräte; Ob dein Modell auch dabei ist, kann ich nicht sagen):
https://github.com/crycode-de/ioBroker.canbus/blob/master/well-known-messages/configs/rotex-hpsu-latest.json
https://github.com/zanac/pyHPSU/blob/master/HPSU/canemu.py

Beide Projekte fragen auch `0xFA 0x09 0x20` für  `EHS CH` ab.
Eine Sonderbehabdlung für irgendwelche Modelle haben sie auch nicht.

Ich tippe darauf, dass deine Pumpe da etwas krumm macht. Genau weiß ich es aber nicht.

wrfz, 2024-11-01:
> ja, wo und wie kann ich eine andere Log erzeugen?

Andere Logs gibt es nicht

Steinmetz1957, 2024-11-01:
Hallo wrfz, ich habe die fehlerhaften Werte in der PDF jetzt Rot und Fett markiert. Der Wert in grüner Schrift ist OK. 
Mich wundert das die Anlage in der Gerätestatistik auf dem Display mir richtige Werte anzeigt ca.37000 bzw 47000 KWh und über den Bus Werte die ich auch keinem anderen möglichen Wert zuordnen kann. selbst mathematisch ist das nicht OK denn 5000 +9904 = nicht 14848. Das lässt zumindest schließen das es wohl 3 Einzelwerte gibt die die Anlage intern speichert. Aber wo?
Danke 
[logs_daikin-rotex_logs.pdf](https://github.com/user-attachments/files/17603571/logs_daikin-rotex_logs.pdf)

wrfz, 2024-11-02:
Ich kann dir leider nicht mehr dazu sagen. Alle Werte aus deinem Log kommen so von der Pumpe.

TNTLarsn, 2025-04-15:
Das ist interessant. Bei mir ist (erst) der Wert für `total_electrical_energy` Elektrische Energie Gesamt betroffen, da die anderen Werte (noch?) unter 32768 KW/h liegen. Allerdings bleibt `total_electrical_energy` bei 32768 KW/h konstant.
Ich muss mal am Gerät prüfen, wieviel es tatsächlich sind

TNTLarsn, 2025-04-15:
"Elektrische Energie Gesamt" im Menü der Anlage nicht gefunden. Alle anderen Energiewerte stimmen.

wrfz, 2025-04-15:
Ich kann dir leider dazu nichts sagen.
So wie die Werte per CAN-Kommen, so werden sie im ESP verarbeitet.
Ein Fehler im Projekt code kann man fast ausschließen.

TNTLarsn, 2025-04-17:
> > Wo gibt es eigentlich eine offizielle Quelle für die Hex Adressen und deren Bedeutung oder sind diese ausschließlich rückwärts durch auslesen ermittelt?
> 
> Wie [@Trunks1982](https://github.com/Trunks1982) geschrieben hat, gibt es keine offizielle Protokolldokumentation. Es gibt allerdings zwei Projekte, die seit längerem existieren und von Nutzern verschiedenster Rotex/Daikin Modelle (mit CAN) eingesetzt werden (unter anderem auch Altherma Geräte; ob dein Modell auch dabei ist, kann ich nicht sagen): https://github.com/crycode-de/ioBroker.canbus/blob/master/well-known-messages/configs/rotex-hpsu-latest.json https://github.com/zanac/pyHPSU/blob/master/HPSU/canemu.py
> 
> Beide Projekte fragen auch `0xFA 0x09 0x20` für `EHS CH` ab. Eine Sonderbehandlung für irgendwelche Modelle haben sie auch nicht.
> 
> Ich tippe darauf, dass deine Pumpe da etwas eigen macht. Genau weiß ich es aber nicht.

In einem dieser Projekte wird der Wert Überlauf  mit nit Betracht gezogen:

https://github.com/crycode-de/ioBroker.canbus/blob/119510e328649a8de2508069aafb3e57ddd100b1/well-known-messages/configs/rotex-hpsu-latest.json#L997

wrfz, 2025-04-17:
@TNTLarsn Danke für den Hinweis! Habe übersehen.

In meinem Fall scheint das Ergebnis keinen Sinn zu machen.

Ich habe folgendes probiert:

1. Zuerst den **Filter** gesetzt, so dass nur noch die Ausgaben des Multiplikator per CAN geloggt werden
2. **Request** an die HPSU gesendet, den Multiplikator zu liefern (**Achtung!** Hier kann man die Pumpe kaputt machen, falls man etwas falsches wegschickt!)
[image]

Das führt bei mir in den Logs zu folgenden Ausgaben (im ESP-WEB-UI):
```
[10:32:33][I][daikin_rotex_can:146]: millis: 653359804|custom_request() can_id<0x680> data<31 00 FA C2 EE 00 00> str<0x31 0x00 0xFA 0xC2 0xEE 0x00 0x00>
[10:32:33][I][unhandled:146]: millis: 653359834|can_id<0x180> data<D2 00 FA C2 EE 80 00>
```

Wie man der zweiten Zeile entnehmen kann, ist das Multiplikator-Byte bei mir **0x80**, das ich von der HPSU zurück bekomme.

Der ioB-code macht folgendes damit (nimmt den Multiplikator mutipliziert mit 32768 und addiert ihn auf den eigentlichen Wert):

```
if (sharedData.fac2ee) {
        value += 32768 * sharedData.fac2ee;
}
```

Ich hatte bei meiner HPSU im Winter einen Factory-Reset gemacht. Deshalb habe ich als Energie Gesamt auf dem Display eine **4.264 kWh**

Per CAN kommt derselbe Wert zurück. Da ich kein Überlauf habe, müsste eigentlich eine **0x01** zurück kommen und nicht eine **0x80**.

Kann einer von euch probieren, den Logfilter zu setzen und den Request zu senden und dann schauen, was im Log zurück kommt?

TNTLarsn, 2025-04-17:
```
12:27:33[I][daikin_rotex_can:146]millis: 20034789|custom_request() can_id<0x680> data<31 00 FA C2 EE 00 00> str<0x31 0x00 0xFA 0xC2 0xEE 0x00 0x00>
12:27:33[I][unhandled:146]millis: 20034810|can_id<0x180> data<D2 00 FA C2 EE 80 00>
```

Sprechen wir eigentlich vom gleichen Wert???

[image]

wrfz, 2025-04-17:
> Sprechen wir eigentlich vom gleichen Wert???

Ja, es geht um **Erzeugte Energie Gesamt** (`FA 09 30`). 
Nur dieser Wert hat laut der ioB-Konfiguration das zusätzliche Register `FA C2 EE`.

Andere Werte (`EHS für DHW`, `EHS fuer CH`, `Energie Kühlung`, `Energie Heizung`, `Energie für WW` und `Elektrische Energie Gesamt`) haben das nicht.

wrfz, 2025-04-17:
> FA C2 FA habe ich im Projekt https://github.com/crycode-de/ioBroker.canbus gerade nicht gefunden... Wo ist dieses Command eigentlich Dokumentiert?.. konnte es bislang in keinem anderen Projekt finden?! Und am Gerät selbst wüsste ich nicht, wo ich es anzeigen kann...

Finde es auch weder bei ioB noch by pyhpsu. Bei meiner HPSU Compact 508 (v4) kommt auch nur `32.762` das enspricht `0x8000` und bedeutet wohl, dass der Wert nicht unterstüzuz wird. Hab auch nichts dazu im Display.

Die CAN-Nachricht habe ich von einem Ultra („klein“ 2018-2021) - Besitzer aus der Telegramgruppe.

> Wo ist dieses Command eigentlich Dokumentiert?

Die CAN-Nachrichten werden von der HPSU über CAN automatisch gesendet, sobald du im Display in das entsprechende Menü rein gehst. So haben die ganzen Projekte ihre Listen aufgebaut.

TNTLarsn, 2025-04-17:
> Ja, es geht um **Erzeugte Energie Gesamt** (`FA 09 30`). Nur dieser Wert hat laut der ioB-Konfiguration das zusätzliche Register `FA C2 EE`.
>
> Andere Werte (`EHS für DHW`, `EHS fuer CH`, `Energie [...]

Dann könnte man den Korrekturwert für *Erzeugte Energie Gesamt** (`FA 09 30`). vielleicht im Projektkode mit einbauen (für alle deren WP die 32768 überschreitet.

Für **Elektrische Energie Gesamt** (`FA C2 FA`) müsste man vielleicht weiter recherchieren.

wrfz, 2025-04-17:
Würde ich ja machen, weiß nur nicht, wie ich den Wert auswerten muss.
Dein Log bringt mich leider nicht weiter, da bei dir `FA C2 EE` auch eine `0x80` liefert.
Ich brächte Logs von Leuten, die bereits einen Überlauf hatten.

wrfz, 2025-04-17:
Alles was ChatGPT ausgespuck hat, war mir bewusst.
Womit ich ein Problem habe, ist die `0x80`.

Machen wir mein konkretest Beispiel:

```
value = 4264; // mein aktuelle "Erzeugte Energie Gesamt"
sharedData.fac2ee = 0x80; // 0x80 als Dezimalwert ergibt 128

if (sharedData.fac2ee) {
    value += 32768 * sharedData.fac2ee;
}
```
Wenn wir nun das ganze ausrechnen: 4264 + 32768 * 128 = 4198568 kWh = 4198 MWh = 4 GWh
Das ist nicht realistisch. So viel Energie kann eine LWP nicht verbrauchen.
Der Wert 0x80 bzw. 128 hat sicherlich irgendeine Bedeutung. Es kann nicht sein, dass meine und dein LWP 128 mal einen Überlauf hatten.

wrfz, 2025-04-17:
Hallo @crycode-de,

wir diskutieren hier über deinen "Erzeugte Energie Gesamt" - Fix:
https://github.com/crycode-de/ioBroker.canbus/commit/69809e9c605c00155a48e4b4e7ea0fee9569af1c

- Kannst mir erklären, was der Wert `0x80` für eine Bedeutung hat?
In deinem Code gibt es dafür keine Sonderbehandlung.
- Wie bist du auf die CAN-Adresse `0xFA 0xC2 0xEE` gekommen?
Ich frage mich, ob es bei den anderen Energie-Werten (`EHS für DHW`, `EHS fuer CH`, `Energie Kühlung`, `Energie Heizung`, `Energie für WW` und `Elektrische Energie Gesamt`) auch jeweils eine Überlaufadresse gibt (die du bei dir nicht implementiert hast).

wrfz, 2025-04-17:
@crycode-de Danke für die Infos!

closed 2025-05-15 by Steinmetz1957