Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #73: Installation in Home Assistant für absoluten Anfänger

giulup, 2025-01-23:
Hallo,

tut mir Leid für diese dumme Frage jetzt aber ich bekomme es einfach nicht hin.

Wenn ich den ESP über die Webseite mit der bin-Datei flashe funktioniert es. Ich habe dann den ESP im Netzwerk  und kann auf ihn zugreifen, aber ich bekomme ihn nicht in mein ESPhome innerhalb Home Assistant integriert weil mir der Verschlüsselungsschlüssel fehlt.

Wenn ich innerhalb von ESPhome in Home Assistant flashen will bekomme ich es nicht hin. Ich weiß da nicht wie ich die bin-Datei hinzufügen kann oder welchen Code ich wohin kopieren muss. Ich bekomme nur Fehlermeldungen wenn ich die Codes aus dem example Ordner benutze

Trunks1982, 2025-01-23:
Du musst dich doch nur an die Anleitung halten wo beschrieben ist über esphome flashen.

giulup, 2025-01-23:
Das habe ich versucht. In der Anleitung steht für die Home Assistant Installation bei

Schritt 2:

4. Lade die Konfigurationsdatei herunter, die für das einmalige Flashen benötigt wird.

Aber danach nicht mehr wo sie verwendet wird.

Habe es gerade vor ein paar Sekunden wie im anderen issue probiert. Jetzt funktioniert es. Vielleicht kannst du die Anleitung an der Stelle noch etwas weiter ausführen. Hab den kompletten code aus der full.yaml genommen, in ESPhome kompiliert und über die Webseite geflasht. So hat es dann funktioniert. Aber direkt aus ESPHome hat es mir immer nur Fehler ausgespuckt. Sehr merkwürdig.

closed 2025-01-23 by giulup