Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #81: Compilation errors

TheNoltens, 2025-02-13:
Hi,
maybe I'm too stupid, but when I compile your full_de example I get an error that millis was not declared in this scope.
I'm quite new in that things and I don't know how to handle this.
Can you help me out?
Attached you'll find my logfile from ESPHome Build

[Log.txt](https://github.com/user-attachments/files/18789112/Log.txt)

So, I see there two issues.

`src/esphome/components/daikin_rotex_can/daikin_rotex_can.h:51:62: error: no matching function for call to 'std::__cxx11::list<std::pair<std::function<void()>, unsigned int> >::push_back(<brace-enclosed initializer list>)'`

`src/esphome/components/daikin_rotex_can/daikin_rotex_can.h:50:36: error: 'millis' was not declared in this scope`

Cicatr1x, 2025-02-21:
Can you share your yaml File?

wrfz, 2025-02-21:
Flash bitte den `bugfix/millis` Branch.
Dazu in deiner yaml den folgenden Teil ändern:

1) Obere `external_components` auskommentieren
2) Untere `external_components` einkommentieren, und `dev` unter `ref` durch `bugfix/millis` ersetzen

[image]

wrfz, 2025-02-21:
You are using `ESPHome 2024.12.4`. The current version is `2025.2.0`. Please update and flash again.

TheNoltens, 2025-02-21:
I've tried it with the current version 2025.2.
Here is my yaml file. It is based on the example that Trunks published here.

[rotex.yaml.txt](https://github.com/user-attachments/files/18912372/rotex.yaml.txt)

wrfz, 2025-02-21:
Ist ESPHome part of HA? Or are you using a standalone ESPhome?

wrfz, 2025-02-21:
Have you tried to clean cache?

[image]

TheNoltens, 2025-02-21:
I‘m using the ESPHome Plugin for HA.
Tomorrow I will try to clean the cache

TheNoltens, 2025-02-22:
> Have you tried to clean cache?
> 
> [image]

I did it that way, but still the same. :-(

wrfz, 2025-02-24:
Sorry, I have no idea. This is the first time I have encountered this problem.

Alternatively you can try to install ESPHome Standalone on your desktop PC.
Use one of the following scripts:
https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/tree/main/scripts

TheNoltens, 2025-02-24:
It seems that the function millis is a part or arduino. Can’t you declare this function in your code? 
The standalone bin-file is working.
HA ist recognizing the ESP, but asks for an encryption key, as it is solved in issue https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/issues/45.
But also there I can’t see the solution for this. 
Can you provide me the key for your bin-file? I guess, this would solve my issue 😀.

Cicatr1x, 2025-02-24:
I tried that config on a different device and its working. If its part of arduino i dont see a reason why ESPHome in HA isnt fetching that properly.

Trunks1982, 2025-02-24:
Viele andere User haben die yaml nach der Anleitung installiert und dieser Fehler ist nie aufgetreten.Vielleicht mal mit anderer Hardware flashen.

Trunks1982, 2025-02-24:
Bitte den einmal testen:  heqmwzvu7TckUlOZjse5Y2dPSiNR8a5+LZDPmSC3URI=

TheNoltens, 2025-02-25:
> Bitte den einmal testen: heqmwzvu7TckUlOZjse5Y2dPSiNR8a5+LZDPmSC3URI=

No, the key it's not working :-(

TheNoltens, 2025-02-25:
Call me stupid. Somehow I managed to install your full example code on the atom now. 
Here are the steps I did:
- Open web.esphome.io and connect the atom to the computer
- I clicked "Prepare for first use" and set up the wifi after installation
- ESPHome in HA found my device and I took the control
- It took a while and two tries but then I saw my device
- I added all your example code and clicked install wirelessly
- It compiled without errors and is running...

But I don't know if this behavior is the expected one. I still have the ESPHome device visible and my rotex device is shown offline, even when it is connected to wifi.

[image]

wrfz, 2025-02-25:
Steps are correct. Do you see a new device in your WIFI list in your router? I guess you tried a reset of the ESP or unplug and plug it again?

TheNoltens, 2025-02-26:
Actually I see my device and have access to it. I had to add a staticIP. But now the logs are shown.
But for some reason I don't see this device in HA as a device. It is not shown as a discovered device...
I could cry. How can I add this device manually?

TheNoltens, 2025-02-26:
Okay, done. Now it is working. After adding an staticIP, it is working.
Many thanks for your help. :-)

closed 2025-02-26 by TheNoltens

wrfz, 2025-02-27:
Should generally also work with dynamically assigned IP.