Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #86: Frage zur Konfiguration

Smartzeug, 2025-03-11:
Hallo Trunks1982🙂,
riesiges Kompliment für deine Arbeit. 
Ich bin fast mit allem durch und werde es gleich mit meiner HPSU verbinden. 

Eventuell mache ich sogar ein Video dazu und dann wirst du ganz groß erwähnt 😉

Ich habe eine Verständnisfrage:
Nach dem flachen des ESP32 S3 schreibst du, dass man [diese Einträge](https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/blob/main/examples/full_de.yaml) kopieren und dann installieren soll.

Wenn ich das mache bekomme ich irgendwann permanent eine Warnung über 1x Warmwasser. 

[image]

Frage: muss ich aus deiner YAML Datei irgendwas verändern / anpassen? 
Weil ich habe mehr oder weniger gar nichts verändert und einfach nur copy / paste. 

Anbei noch die Logs:
[logs_rotex-hpsu_run.txt](https://github.com/user-attachments/files/19184576/logs_rotex-hpsu_run.txt)

Dankeschön für deine Arbeit und Mühe, bist der Beste!

Trunks1982, 2025-03-11:
Hi dich kenn ich doch:-) 
frage hast du die gpios angepasst oder einmal das Kabel am Can Bus untereinander getauscht?

Smartzeug, 2025-03-11:
Dankeschön für die schnelle Rückmeldung mein Bester. 

Tatsächlich hatte ich einen Zahlendreher drin. 
Hab es nun gerade gezogen und installiere es erneut. 

Jetzt bekomme ich keine Warnung mehr aber unendlich mal eine andere Meldung. 

So ist übrigens die Belegung bei mir 
[image]

Cicatr1x, 2025-03-11:
Du kannst auch den neuen Atom S3 direkt aufs CAN Modul aufsetzen.

https://docs.m5stack.com/en/atom/atom_can

Geh da mal runter zu "Compatible", dann kannst du dir die ganzen Dupont Stecker sparen.

Cicatr1x, 2025-03-11:
@Trunks1982 Bei ihm fehlt aktuell noch der Handle, es werden also Kommandos verschickt aber es gibt noch keine Antwort, die per Handle abgefangen werden.

Trunks1982, 2025-03-11:
https://github.com/user-attachments/assets/a6723c5c-510c-438f-b1a0-52080cdb183a

Smartzeug, 2025-03-11:
Hallo in die Runde🙂
habe den Tipp verfolgt und es direkt auf das CAN Modul gelegt. 

[image]

habe dann in der Konfiguration folgendes eingeben:
`canbus:
  - platform: esp32_can
    id: can_bus
    can_id: 0x680
    tx_pin: GPIO06
    rx_pin: GPIO05
    bit_rate: 20kbps`

Leider kommt der gleicher Fehler. 

[logs_rotex-hpsu_run.txt](https://github.com/user-attachments/files/19187784/logs_rotex-hpsu_run.txt)

Trunks1982, 2025-03-11:
Schliess jetzt erstmal an die rotex an.Kannst ja immer noch über wireless flashen alles ändern.

wrfz, 2025-03-11:
> Leider kommt der gleicher Fehler.
[logs_rotex-hpsu_run.txt](https://github.com/user-attachments/files/19187784/logs_rotex-hpsu_run.txt)

Im Log ist kein Fehler!
*sendGet* sind Requests, die der ESP zyklisch raussendet.
Da der ESP keine Antwort von der Pumpe erhält, versucht der ESP es immer wieder (mit 1 x Warmwasser, da es einfach als erstes in der Liste steht).
Sobald du alles korrekt verdrahtet, entsprechend die GPIOs im yaml konfiguriert hast und die Pumpe dann antwortet,
wirst du im Log Antwortnachrichten sehen die links als *handle* gekennzeichnet sind:

[16:43:04][I][sendGet:146]: 1 x Warmwasser can_id<0x680> command<31 00 FA 01 44 00 00>
[16:43:04][I][handle :146]: 1 x Warmwasser<Aus> can_id<0x180> data<D2 00 FA 01 44 00 00> changed<0>

Versuch mal endweder die CAN-Keitungen zu verstauschen oder rx_pin/tx_pin in deiner yaml.

Smartzeug, 2025-03-11:
Hallo zusammen,
kleines Update 

[image]

[logs_rotex-hpsu_run-2.txt](https://github.com/user-attachments/files/19191097/logs_rotex-hpsu_run-2.txt)

wrfz, 2025-03-11:
Sieht nach einem DNS-Problem aus.
Probier mal die IP statisch im yaml zu definieren:

```
wifi:
  ssid: "Your_SSID"
  password: "Your_Password"
  manual_ip:
    static_ip: 192.168.1.100  # Deine IP
    gateway: 192.168.1.1      # Dein Netz
    subnet: 255.255.255.0
```

DHCP sollte natürlich auch gehen. Kannst später wieder entfernen, wenn du die Ursache gefunden hast. Funktioniert eigentlich meistens ohne Probleme.

Trunks1982, 2025-03-11:
> Hallo zusammen, kleines Update, habe es nun mit der Rolex verbunden. Und wenn ich den wireless flash starte kommt folgendes:
> 
> [image]
> 
> [logs_rotex-hpsu_run-2.txt](https://github.com/user-attachments/files/19191097/logs_rotex-hpsu_run-2.txt)

Hey hast du mal eben Lust in unsren telegram chat zu kommen? Dann können wir das schneller beantworten.

Trunks1982, 2025-03-11:
Ich habe die seid und passoword fest in der secret.yaml eingetragen.Das sollte eigentlich reichen.

wrfz, 2025-03-11:
> Ich habe die seid und passoword fest in der secret.yaml eingetragen.Das sollte eigentlich reichen.

Sein Fehler hat eigentlich nichts mit Passwort, EncryptionKey oder ssid zutun.
ESPhome kann den Name `rotex-hpsu.local` nicht per DNS auflösen:

`INFO Resolving IP address of rotex-hpsu.local in mDNS`
`ERROR Error resolving IP address of rotex-hpsu.local. Is it connected to WiFi?`
Das hatten schon 1-2 Leute.

Smartzeug, 2025-03-11:
> > Ich habe die seid und passoword fest in der secret.yaml eingetragen.Das sollte eigentlich reichen.
> 
> Der Fehler hat nichts mit dem Passwort, dem EncryptionKey oder der SSID zu tun. ESPhome kann den Namen `rotex-hpsu.local` nicht über DNS auflösen.
> 
> `INFO Resolving IP address of rotex-hpsu.local in mDNS` `ERROR Error resolving IP address of rotex-hpsu.local. Is it connected to WiFi?`
> 
> Das hatten schon 1-2 Leute. Ist aber eher eine Ausnahme.

Hab im Unifi mal nachgeschaut und finde dort auch nicht den Atom S3 lite 

[image]

Trunks1982, 2025-03-11:
Entweder du gibst das so ein wie wrz beschrieben hat oder in esphome secret.yaml die ssid und passwort für das wlan ein.Probier mal aus

wrfz, 2025-03-11:
Bin davon ausgegangen, dass er bereits alles in secrets.yaml eingetragen hat.
Sehe allerdings gerade, dass dieser Schritt in der Doku komplett fehlt.

Trunks1982, 2025-03-11:
Werde den punkt morgen mal nachtragen.

Smartzeug, 2025-03-11:
Das hatte ich bereits vorher schon drin durch den anderen ESP :-)
[image]

Trunks1982, 2025-03-11:
Ok das sieht gut aus. Haste bei dem anderen ESp ne feste ip vergeben oder auch so wie im unserer yaml?

wrfz, 2025-03-11:
Der Fehler sagt relativ eindeutig, dass der ESP keine IP bekommt. Kann unteranderem folgende Gründe haben:
- Gerät ist nicht mit WiFi verbunden
- Gerät hat keine IP-Adresse
- mDNS funktioniert nicht richtig
- Router blockiert mDNS

Wie gesagt, teste statisch die IP einzutragen, wie hier beschrieben: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/issues/86#issuecomment-2714921360

Sobald der Router und das Gerät sich bekannt gemacht haben, kannst versuchen die statische IP wieder zu entfernen.

Smartzeug, 2025-03-11:
Kurze Verständnisfrage:
woher bekommt das ESP seinen Strom wenn es am Rolex hängt?

Smartzeug, 2025-03-11:
Es lebt :-)
[image]

Trunks1982, 2025-03-11:
Darf ich fragen was du für eine Rotex hast?

Smartzeug, 2025-03-11:
HPSU 308 6kW

Kurze Frage: wo finde ich die serielle Schnittstelle. Würde gerne nutzen was verbaut ist anstatt ein weiteres USB Kabel dort liegen zu haben

wrfz, 2025-03-11:
Musst die große Platine entfernen. Sitzt auf der Platine dahinter.

wrfz, 2025-03-11:
[image]

wrfz, 2025-03-11:
Falls du auch serielle Daten (Kompressordrehzahl, Lüfterdrehzahl, Kältemitteltemperatur, ...) auslesen willst dann musst da ehe ran.
Infos vom Aussengerät kommen nicht über CAN. Nur seriell.
Dann solltest dir das Projekt hier anschauen: https://github.com/wrfz/esphome-components

Trunks1982, 2025-03-11:
[image]

Würde dann komplett so aussehn😉

closed 2025-03-11 by Smartzeug

Trunks1982, 2025-03-11:
https://t.me/+C1iVhAjaENw0ODY0

Komm mal hier rein.Da haben wir auch noch mehr fotos und regen Austausch über unsere Rotex.

wrfz, 2025-03-11:
Die Entities, die du im ESPhome->yaml verwenden möchtest, solltest du in HA-configuration.yaml so angeben:

```
recorder:
  exclude:
    entities:
      - sensor.hpsu_can_system_date
      - sensor.hpsu_can_system_time
      - number.hpsu_can_system_date_day
      - number.hpsu_can_system_date_month
      - number.hpsu_can_system_date_year
      - number.hpsu_can_system_time_hour
      - number.hpsu_can_system_time_minute
      - number.hpsu_can_system_time_second
```

Verhindert, dass die Uhrzeit - States in HA DB geschrieben werden.

Smartzeug, 2025-03-11:
> Die Uhreit-Entities, die du im ESPhome -> yaml verwenden möchtest (einkommentierst), solltest du in HA -> `configuration.yaml` so angeben:
> 
> ```
> recorder:
>   exclude:
>     entities:
>       - sensor.hpsu_can_system_date
>       - sensor.hpsu_can_system_time
>       - number.hpsu_can_system_date_day
>       - number.hpsu_can_system_date_month
>       - number.hpsu_can_system_date_year
>       - number.hpsu_can_system_time_hour
>       - number.hpsu_can_system_time_minute
>       - number.hpsu_can_system_time_second
> ```
> 
> Verhindert, dass die Uhrzeit - States in HA - DB geschrieben werden.

Dankeschön :-)

wrfz, 2025-03-12:
> Es lebt :-) 

Was war schlussendlich die Ursache? Würden andere sicherlich auch gerne wissen.

Smartzeug, 2025-03-12:
> > Es lebt :-)
> 
> Was war schlussendlich die Ursache? Würden andere sicherlich auch gerne wissen.

Ich habe alles einfach nur installiert. Dann ging es 🙂👍

Smartzeug, 2025-03-12:
Kennt ihr das Phänomen wenn er eigenständig zu Heizen springt?
Ihr fahre eigentlich Automatik 1 aber sobald das endet schaltet es auf Heizen. 

[image]

Trunks1982, 2025-03-12:
Leider nein.Wieso nutzt du Automatik?
Besser wäre es die Rotex durchlaufen zulassen.

wrfz, 2025-03-12:
> Kennt ihr das Phänomen wenn er eigenständig zu Heizen springt? Ihr fahre eigentlich Automatik 1 aber sobald das endet schaltet es auf Heizen.

Ich denke, du hast `Abtauoptimierung` aktiviert. Es sorgt dafür, dass beim Abtauen kurzzeitig auf Sommer geschaltet wird,
um zu vermeiden, dass Warmwasser in den Fußboden gepumpt wird.
Keiner hat den Fehler bisher gemeldet.
Liegt wohl daran, dass alle bisher `Heizen` verwendet haben.

Smartzeug, 2025-03-12:
Ihr habt absolut recht mit dem durchlaufen lassen. 
Das mache ich auch im Kernwinter. 
Aber jetzt wo die Tage länger und die Sonne öfter sich blicken lässt wechsle ich den Modus. 
Durch die PV & Batterie lohnt es sich bei mir nur Tagsüber mit höheren Temperaturen laufen zu lassen als durchgehend.

Ich beobachte das mal weiter.

Smartzeug, 2025-03-12:
> > > > Es lebt :-)
> > > 
> > > 
> > > Was war schlussendlich die Ursache? Würden andere sicherlich auch gerne wissen.
> > 
> > 
> > Ich habe alles einfach nur installiert. Dann ging es 🙂👍
> 
> Ich meinte wie hast du das Problem mit den nicht aufgelösten IP in den Griff bekommen? [#86 (comment)](https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/issues/86#issuecomment-2714921352) Flashen wollte ja zunächst nicht. Hast etwas geändert oder einfach erneut geflasht?

Hab über Web geflasht wie es jemand anderes auch schon in einem anderen issue vorgeschlagen hatte. Danach ging alles nach dem anschließen an die HPSU

wrfz, 2025-03-12:
> Ihr habt absolut recht mit dem durchlaufen lassen. Das mache ich auch im Kernwinter. Aber jetzt wo die Tage länger und die Sonne öfter sich blicken lässt wechsle ich den Modus. Durch die PV & [...]

Mit PV ist es eigentlich besser genau umgekehrt zu machen.
Tagsüber durchlaufen lassen und nachts ausschalten.
Tagsüber kannst mit der neuen Steuerung beliebig die Vorlauftemperaur regeln. 
(Solltest allerdings maximal 1 / Stunde regeln, damit der EEPROM der HPSU nicht irgendwann den Geist aufgibt)

Trunks1982, 2025-03-12:
> Ihr habt absolut recht mit dem durchlaufen lassen. Das mache ich auch im Kernwinter. Aber jetzt wo die Tage länger und die Sonne öfter sich blicken lässt wechsle ich den Modus. Durch die PV & [...]

Dafür haste ja jetzt die möglichkeit ne automatiserung zu bauen in HA. Also Automatik würde ich an deiner Stelle jetzt nichtvmehr nutzen.

Smartzeug, 2025-03-12:
> Mit PV ist es eigentlich besser genau umgekehrt zu machen. Tagsüber durchlaufen lassen und nachts ausschalten. Tagsüber kannst mit der neuen Steuerung beliebig die Vorlauftemperaur regeln. (Solltest [...]

Genau das habe ich davor gemacht. Die Automatik heizt morgens bis abends. Eigentlich sollte es jetzt aus gehen. Aber es springt auf Heizen. 
Deshalb meine Frage in die Runde 🙂

Hab die Automatisierung bereits gebaut und teste es gerade.

wrfz, 2025-03-12:
> Kennt ihr das Phänomen wenn er eigenständig zu Heizen springt? Ihr fahre eigentlich Automatik 1 aber sobald das endet schaltet es auf Heizen.

Hab den Fehler gefixt. Werde voraussichtlich in ein paar Tagen releasen.