Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #87: Rotex HPSU Bi-Bloc keine CAN Verbindung

Michl09, 2025-03-19:
Hallo,

Rotex RHBX08CB9W
Baujahr Februar 2016

auf der Hauptplatine A1P Stecker X10 konnte ich vor Jahren bereits mit einem DSD SH-U09C3 den Can Bus auslesen.
Pinbelegung X10:
Pin 1: VCC
Pin 2: RXD
Pin 3: TXD
Pin 4: Steuerspannung für LED
Pin 5: GND

[image]

Jetzt habe ich einen AtomS3U mit einer Unit Mini Can angeschlossen. Ich habe auch Pin 2 und 3 mal getauscht.

[image]

Leider kann ich kein Signal von der Wärmepumpe empfangen. Hier ein Ausschnitt aus den Logs. Das wiederholt sich fortwährend:

INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/esphome-web-b6325c.yaml...
INFO [Translate] All translations are consistent!
INFO Project Git Hash d500909
INFO Detected timezone 'Europe/Berlin'
INFO Starting log output from 192.168.178.39 using esphome API
INFO Successfully connected to hpsu-can @ 192.168.178.39 in 0.005s
INFO Successful handshake with hpsu-can @ 192.168.178.39 in 0.062s
[12:42:32][I][app:100]: ESPHome version 2025.2.2 compiled on Mar 19 2025, 09:27:51
[12:42:33][I][sendGet:146]: 1 x Warmwasser can_id<0x680> command<31 00 FA 01 44 00 00>
[12:42:33][W][component:237]: Component daikin_rotex_can took a long time for an operation (1005 ms).
[12:42:33][W][component:238]: Components should block for at most 30 ms.
[12:42:37][I][sendGet:146]: 1 x Warmwasser can_id<0x680> command<31 00 FA 01 44 00 00>
[12:42:37][W][component:237]: Component daikin_rotex_can took a long time for an operation (1009 ms).
[12:42:37][W][component:238]: Components should block for at most 30 ms.

Wo könnte der Fehler liegen?

Viele Grüße

Trunks1982, 2025-03-19:
Und nur h und l belegen.Die anderen brauchste nicht.

wrfz, 2025-03-19:
Versuchst du gerade deine HPSU zu grillen? Oder warum schließt du das CAN-Modul an die Serielle Schnittstelle der HPSU an?

Michl09, 2025-03-19:
ok, abgeklemmt ist jetzt. 
Den J13 Stecker habe ich nicht.  Ich kann nur an den X10 Stecker. Wie könnte ich anklemmen?

wrfz, 2025-03-19:
Zur Klarstellung:
Die HPSU hat zwei Schnittstellen:
- CAN-Schnittstelle: Liefert Infos zum Heizkreis und erlaubt das Gerät zu steuern.
- Serielle Schnisstelle: Liefert Kältekreis-Infos (Kompressordrehzahl, Lüfterdrehzahl, Temperatur des Kältemittels, ...)

Was ist dein Ziel? Welche Werte willst den haben?
CAN-Modul brauchst du nur, wenn du per CAN kommunizieren willst.

Um die Serielle Daten zu lesen, schliesst du den ESP S3 direkt an die X10A Schnittstelle.

Zur Info für die anderen, die das hier lesen:
Der ESP, den du hast kannst direkt anschliessen. Die meisten anderen ESPs vertragen dagegen keine 5 Volt!

wrfz, 2025-03-19:
Um CAN-Daten zu lesen und die HPSU zu steuern, musst du dein ESP mit dem CAN-Modul hier anschliessen:
[image]
Das Projekt, das du auf den ESP flashst ist das hier:
https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/

wrfz, 2025-03-19:
Um Kältekreis-Daten seriell zu lesen, musst du deinen ESP direkt (ohne CAN-Modul!) an die `X10A` Schnittstelle anschliessen:
[image]
Das Projekt dafür findest du hier:
https://github.com/wrfz/esphome-components

Michl09, 2025-03-19:
An der Bedieneinheit RKRUCBL1 verstelle ich momentan manuall die Heizkreis - Vorlauftemperatur.  Die Bedieneinheit ist hier angeschlossen:

[image]

Die Heizkreisvorlauftemperatur würde ich gerne über den ESP32S3 und Homeassistant ändern können. Gibt es die Möglichkeit dann überhaupt?

wrfz, 2025-03-19:
Mit dem CAN-Projekt kannst neben der Vorlauftemperatur nahezu alles ändern, was du auch direkt am Control-Panel der HPSU ändern kannst.

Einfach mal Readme der beiden Projekte gründlich durchlesen. Steht alles relativ detailiert beschrieben:
- https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell
- https://github.com/wrfz/esphome-components

wrfz, 2025-03-19:
Sorry, habe nicht darauf geachtet, dass du eine `ROTEX HPSU NiedertemperaturBibloc` hast.
Wie ich sehe, haben die kein CAN. Dann wirst sie leider nicht steuern können.

wrfz, 2025-03-19:
Bei `HPSU Compact` kommuniziert die `RoCon BM1` mit dem `RoCon B1` über `CAN`. Sieht man oben auf dem Bild.
Da dein Gerät kein `RoCon BM1` hat, wird es `CAN` eher nicht sein. Ist vermuttlich ein anderes serielles Protokoll.
Mehr kann ich dazu leider nicht sagen.

Michl09, 2025-03-19:
Auch wenn es jetzt für mich nicht zum Ziel geführt hat, dankeschön für die Hilfe und Unterstützung. Ich konnte viel dazu lernen.