Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #88: Cannot Compile anymore

TNTLarsn, 2025-03-27:
Hi there,

```log
INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/rotex.yaml...
INFO [Translate] All translations are consistent!
INFO Project Git Hash 34c9b7e
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO [Translate] Setting language to 'de'
INFO Writing cpp translate file
INFO Building cpp translate dictionary for language: de
INFO Generated /data/external_components/5f1202b0/components/daikin_rotex_can/translations.cpp
INFO Compiling app...
Processing rotex (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Linking .pioenvs/rotex/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp.o: in function `esphome::daikin_rotex_can::DaikinRotexCanComponent::recalculate_state(esphome::EntityBase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/data/build/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:541: undefined reference to `void esphome::daikin_rotex_can::Utils::log<float, float, float, float, float, unsigned long, unsigned long>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float, float, float, float, float, unsigned long, unsigned long)'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/build/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:556: undefined reference to `void esphome::daikin_rotex_can::Utils::log<float, float, float, float, float, unsigned long, unsigned long>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, float, float, float, float, float, unsigned long, unsigned long)'
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: /data/build/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:584: undefined reference to `void esphome::daikin_rotex_can::Utils::log<char const*, bool, float, float, bool, unsigned long, unsigned long>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, bool, float, float, bool, unsigned long, unsigned long)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/rotex/firmware.elf] Error 1
========================= [FAILED] Took 14.21 seconds =========================
```

maybe something went wrong during: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/commit/26ccd21739202ed84e6f21bb4ea5c3f8c7acb147

My yaml just worked until the latest Update of esphome, and compiling never failed so far.
If you think the yaml is relevant I will post it.

Trunks1982, 2025-03-27:
Nutzt du den esp miitels zigbee?

Ich hab gerade mal mit dem neuesten esphome nen update gemacht ohne probleme.

TNTLarsn, 2025-03-27:
No, I´m using WLAN.
Here's the yaml file

```yaml
substitutions:
  name: "rotex"
  friendly_name:  WiCan Rotex
  tx_pin: '0'
  rx_pin: '3'
  ip_address: "192.168.178.156"

esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

esphome:
    name: "${name}"
    friendly_name: "${friendly_name}"
    platformio_options:
        build_unflags:
            - "-std=gnu++11"
            - "-fno-rtti"
        build_flags:
            - "-std=gnu++17"

external_components:
  - source: github://Trunks1982/Daikin-Rotex-HPSU-CAN

#external_components:   # Use this block instead of the one above if you want to use the development branch
#  - source:
#      type: git
#      url: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN.git
#      ref: dev
#    refresh: 0s

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "O2q/wKyy+6/blNRgkwoT+a3Rb+oy2Zp4POd4BTpRJWk="

ota:
  - platform: esphome
    password: !secret ota_password

web_server:
  version: 3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ### Trying to change the name of a node or its address in the network? 
  ### You can do so with the use_address option of the WiFi configuration.
  ### Change the device name or address in your YAML to the new value and additionally set use_address to point to the old address like so:

  ### Remove or comment out use_address
  ### use_address: ${ip_address}

  ### Now upload the updated config to the device. 
  ### As a second step, you now need to remove the use_address option from your configuration again so that subsequent uploads will work again
  ### (otherwise it will try to upload to the old address).

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} Fallback"
[... 402 lines cut ...]
    thermal_power:
      name: Thermische Leistung
    optimized_defrosting:
      name: Abtau-Optimierung
    temperature_spread:
      name: Spreizung
    tv_tvbh_delta:
      name: Tv-TVBH Delta
    tvbh_tr_delta:
      name: TVBH-Tr Delta
    vorlauf_soll_tv_delta:
      name: VorlaufSoll-Tv Delta

# Rocon U1

    #t_room:
    #  name: Raum-Ist

# To be categorized

    target_supply_temperature:
      name: Vorlauf Soll
    supply_setpoint_regulated:
      name: Vorlauf Soll Geregelt
```

Cicatr1x, 2025-03-27:
Mach mal vor dem nächsten Build ein "Clean Build Files" in ESPhome.

TNTLarsn, 2025-03-27:
> Mach mal vor dem nächsten Build ein "Clean Build Files" in ESPhome.

already done - no success (I did read https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/issues/81 beforehand)

wrfz, 2025-03-27:
Please comment out the `main` branch and comment in `dev` branch like this:
```
#external_components:
#  - source: github://Trunks1982/Daikin-Rotex-HPSU-CAN

external_components:   # Use this block instead of the one above if you want to use the development branch
  - source:
      type: git
      url: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN.git
      ref: dev
    refresh: 0s
```

TNTLarsn, 2025-03-27:
just saw it: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/commit/d31cfc12a33a803eb8978e4f54bacb824e949bf6 

please also add:

```c++
template void Utils::log<char const*, bool, float, float, bool, unsigned long, unsigned long>(std::string const&, std::string const&, char const*, bool, float, float, bool, unsigned long, unsigned long);
```

then ist working.

wrfz, 2025-03-27:
please share the current error log

TNTLarsn, 2025-03-27:
Here's **without** my pull request

````
INFO ESPHome 2025.3.2
INFO Reading configuration /config/esphome/rotex.yaml...
INFO Updating https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN.git@dev
INFO [Translate] All translations are consistent!
INFO Project Git Hash d31cfc1
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO [Translate] Setting language to 'de'
INFO Writing cpp translate file
INFO Building cpp translate dictionary for language: de
INFO Generated /data/external_components/68c06f53/components/daikin_rotex_can/translations.cpp
INFO Compiling app...
Processing rotex (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.4
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.3.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.6
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/rotex/src/esphome/components/daikin_rotex_can/utils.cpp.o
Compiling .pioenvs/rotex/src/main.cpp.o
Linking .pioenvs/rotex/firmware.elf
/data/cache/platformio/packages/toolchain-riscv32-esp@8.4.0+2021r2-patch5/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: .pioenvs/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp.o: in function `esphome::daikin_rotex_can::DaikinRotexCanComponent::recalculate_state(esphome::EntityBase*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
/data/build/rotex/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:588: undefined reference to `void esphome::daikin_rotex_can::Utils::log<char const*, bool, float, float, bool, unsigned long, unsigned long>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, bool, float, float, bool, unsigned long, unsigned long)'
collect2: error: ld returned 1 exit status
*** [.pioenvs/rotex/firmware.elf] Error 1
========================= [FAILED] Took 53.76 seconds =========================
````

wrfz, 2025-03-27:
Please try to flash

TNTLarsn, 2025-03-27:
working now. You may commit to master.
Thank you very much

wrfz, 2025-03-27:
It's interesting that it worked for me, for the Github pipeline and also for some other users, but not for you.
Which ESP are you using?

TNTLarsn, 2025-03-27:
https://www.meatpi.com/products/usb-dual-can

ESP32C3

Cicatr1x, 2025-04-04:
> Hello,
> 
> compilation Error, my config is very simple:
> 
> esp32: board: esp32-s3-devkitc-1 framework: type: arduino
> 
> external_components:
> 
> * source: github://Trunks1982/Daikin-Rotex-HPSU-CAN
> 
> canbus:
> 
> * platform: esp32_can
>   id: can_bus
>   can_id: 0x680
>   rx_pin: GPIO5
>   tx_pin: GPIO6
>   bit_rate: 20kbps
> 
> daikin_rotex_can: language: en id: rotext_hpsu canbus_id: can_bus project_git_hash: name: Project Git Hash entities: tdhw1: name: Hot Water Temperature
> 
> Error compiling:
> 
> Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/utils.cpp.o In file included from src/esphome/components/daikin_rotex_can/entity.cpp:1: src/esphome/components/daikin_rotex_can/entity.h: In member function 'bool esphome::daikin_rotex_can::TEntity::isChangedInLastMilliseconds(uint32_t) const': src/esphome/components/daikin_rotex_can/entity.h:128:16: error: 'millis' was not declared in this scope return millis() > (getLastValueChange() + milliseconds); ^~~~~~ src/esphome/components/daikin_rotex_can/entity.h:128:16: note: suggested alternative: 'killpg' return millis() > (getLastValueChange() + milliseconds); ^~~~~~ killpg src/esphome/components/daikin_rotex_can/entity.h: In member function 'bool esphome::daikin_rotex_can::TEntity::isGetNeeded() const': src/esphome/components/daikin_rotex_can/entity.h:195:37: error: 'millis' was not declared in this scope return getLastUpdate() == 0 || (millis() > (getLastUpdate() + update_interval));
> 
> Ed

Please dont forget to configure the correct platformio options.

esphome:
    name: hpsu-can
    friendly_name: HPSU.CAN
    platformio_options:
        build_unflags:
            - "-std=gnu++11"
            - "-fno-rtti"
        build_flags:
            - "-std=gnu++17"

Bl00d-B0b, 2025-04-06:
Hello,

Sorry to mention, but I try both with and without platform_options. and I get same result:"

`esphome:
  name: $devicename
  friendly_name: $long_devicename
  platformio_options:
    build_unflags:
    - "-std=gnu++11"
    - "-fno-rtti"
    build_flags:
    - "-std=gnu++17"
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

external_components:      # Automatically loads the *esphome-daikin-rotex-can* component from github during compilation
  - source: github://Trunks1982/Daikin-Rotex-HPSU-CAN
`

Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/entity.cpp.o
Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/entity_manager.cpp.o
Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/pid.cpp.o
Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/sensors.cpp.o
Compiling .pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/texts.cpp.o
In file included from src/esphome/components/daikin_rotex_can/sensors.h:5,
                 from src/esphome/components/daikin_rotex_can/entity_manager.h:3,
                 from src/esphome/components/daikin_rotex_can/daikin_rotex_can.h:4,
                 from src/esphome/components/daikin_rotex_can/buttons.h:3,
                 from src/esphome/components/daikin_rotex_can/buttons.cpp:1:
src/esphome/components/daikin_rotex_can/entity.h: In member function 'bool esphome::daikin_rotex_can::TEntity::isChangedInLastMilliseconds(uint32_t) const':
src/esphome/components/daikin_rotex_can/entity.h:128:16: error: 'millis' was not declared in this scope
         return millis() > (getLastValueChange() + milliseconds);
                ^~~~~~
src/esphome/components/daikin_rotex_can/entity.h:128:16: note: suggested alternative: 'killpg'
         return millis() > (getLastValueChange() + milliseconds);
                ^~~~~~
                killpg

wrfz, 2025-04-13:
> Sorry to mention, but I try both with and without platform_options. and I get same result:"

It is not clear to me why the **millis** function is not found.
Have a look here: https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/issues/81

@TheNoltens had the same problem and was apparently able to resolve the issue after reflashing the ESP via **web.esphome.io**.

Basically it should work to flash the following yaml:
https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell/blob/main/examples/full_de.yaml

wrfz, 2025-04-13:
> working now. You may commit to master. Thank you very much

@TNTLarsn Please close the topic as your problem has been solved.

@Bl00d-B0b Please create a new issue for your problem.

closed 2025-04-13 by TNTLarsn