Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #90: negative thermische Leistung beim Heizen

Steinmetz1957, 2025-04-02:
Ich habe vorgestern von Version 1.57 auf die aktuelle Version mittels OTA geupdatet. Heute fällt mir auf das mir eine negative thermische Leistung ausgewiesen wird obwohl ich noch heize und der Kompressor aus ist. Außerdem steht die Außentemperatur rund um die Uhr auf 4.5 ° obwohl T-Außen über 16 ° anzeigt. Wenn die Anlage ins Standby wechselt wegen zu hoher Außentemperatur zeigt mir die Durchflußmenge etwas von über 65000 L/Std.
Kann mir jemand sagen ob meine Heizung spinnt oder die Software.

[image]

wrfz, 2025-04-02:
> Ich habe vorgestern von Version 1.57 auf die aktuelle Version mittels OTA geupdatet. Heute fällt mir auf das mir eine negative thermische Leistung ausgewiesen wird obwohl ich noch heize und der Kompressor aus ist.
Eine negative thermische Leistung ist rein logisch möglich. Die thermische Leistung wird aus der Temperaturdifferenz zwischen Vorlauf (Tv) und Rücklauf (Tr) – auch Spreizung genannt – sowie dem Durchfluss berechnet. Je größer die Spreizung oder der Durchfluss, desto mehr Wärme wird im Kondensator (Wärmetauscher) erzeugt.

In bestimmten Situationen, etwa beim Abtauen, kann die Spreizung negativ werden. Das kann auch passieren, wenn der Kompressor abschaltet und die Pumpe die Ventilstellung verändert. Wird die Spreizung negativ, bedeutet das, dass nicht der Kondensator das Wasser erwärmt, sondern das Wasser den Kondensator (bzw. das Kältemittel).

Da in diesem Fall dem Wasserkreislauf thermische Energie entzogen wird, ist eine negative thermische Leistung mathematisch korrekt.

wrfz, 2025-04-02:
| Außerdem steht die Außentemperatur rund um die Uhr auf 4.5 ° obwohl T-Außen über 16 ° anzeigt.

Da müsste man sich die ESP-Logs anschauen, was deine Pumpe für Werte liefert. Ist auf jeden Fall sehr ungewöhlich.

wrfz, 2025-04-02:
> Heute kann das mit dem Abtauen aber nicht zutreffen bei 16° Außentemperatur.

Das war nur ein Bespiel. Hab dir auch ein anderes genannt. 
Dazu müsstest du HA-Kennlinien liefern. Sonst kann ich nichts dazu sagen.

Mindestens diese hier: 
- Betriebsart
- Vorlauf
- Rücklauf
- Durchfluss

Steinmetz1957, 2025-04-02:
Die Werte sind doch auf dem Screenshot zu sehen.

wrfz, 2025-04-02:
> Wenn die Anlage ins Standby wechselt wegen zu hoher Außentemperatur zeigt mir die Durchflußmenge etwas von über 65000 L/Std.

Hast du einen externen Außentemperatursensor verbaut? In diesem Fall wechselt die Pumpe in den ES-Mode (Energiesparmodus).

Der Wert 65.000 entspricht dem Hexadezimalwert 0xFDE8. Wahrscheinlich dient das als Kennzeichnung der Pumpe, dass der Wert nicht mehr aktualisiert wird und ungültig ist. Das deutet nicht auf einen Defekt hin – Rotex hat es einfach so implementiert.

Steinmetz1957, 2025-04-02:
nein es gibt keinen externen Sensor.

Steinmetz1957, 2025-04-02:
wie kann ich das darstellen?

Steinmetz1957, 2025-04-03:
Also die ganzen Fehler müssen mit einem fehlerhaften Update zusammenhängen. Ich habe das Update noch einmal über die yaml durchgeführt und jetzt werden wieder realistische Werte angezeigt, auch die Außentemperatur stimmt wieder.

closed 2025-04-03 by Steinmetz1957