Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #92: Sensor T-WE <t_hs> out of range

TNTLarsn, 2025-04-15:
From the logs
```
[E][daikin_rotex_can:042]
CanSensor::handleValue() => Sensor<t_hs> hex<0xFE0C> uint16<65036> float<6503.600098> out of range[1.000000, 90.000000]
```

TNTLarsn, 2025-04-15:
```
INFO ESPHome 2025.3.3
INFO Reading configuration /config/esphome/rotex.yaml...
INFO Updating https://github.com/Trunks1982/Daikin-Rotex-HPSU-CAN.git@dev
INFO [Translate] All translations are consistent!
INFO Project Git Hash fd11e6e
INFO Detected timezone 'Europe/Berlin'
INFO Starting log output from 192.168.4.234 using esphome API
INFO Successfully connected to rotex @ 192.168.4.234 in 0.032s
INFO Successful handshake with rotex @ 192.168.4.234 in 0.075s
[14:30:15][I][app:100]: ESPHome version 2025.3.3 compiled on Apr 14 2025, 12:02:20
[14:30:15][I][sendGet:146]: millis: 95172573|Fehlercode can_id<0x680> command<31 00 FA 13 88 00 00>
[14:30:15][I][error code:146]: millis: 95172592|tv: 38.900002, tvbh: 39.799999, TvBH-Tv: 0.300000, dhw: 0.000000, flow: 0.000000, dhw_ts: 0, millis: 95172591
[14:30:15][I][error code:146]: millis: 95172595|tvbh: 39.799999, tr: 36.700001, Tr-TvBH: 0.300000, bpv: 100.000000, flow: 0.000000, bpv_ts: 0, millis: 95172594
[14:30:15][I][handle :146]: millis: 95172604|Fehlercode<Kein Fehler> can_id<0x180> data<D2 00 FA 13 88 00 00> changed<0>
[14:30:15][I][sendGet:146]: millis: 95172607|Ruecklauftemperatur Heizung can_id<0x680> command<31 00 FA C1 00 00 00>
[14:30:15][I][handle :146]: millis: 95172630|Ruecklauftemperatur Heizung<36.700001> can_id<0x180> data<D2 00 FA C1 00 01 6F> changed<0>
[14:30:15][I][sendGet:146]: millis: 95172635|Heizkreis Vorlauf (TV) can_id<0x680> command<31 00 FA C0 FC 00 00>
[14:30:15][I][handle :146]: millis: 95172657|Heizkreis Vorlauf (TV)<38.900002> can_id<0x180> data<D2 00 FA C0 FC 01 85> changed<0>
[14:30:15][I][sendGet:146]: millis: 95172660|T-WE can_id<0x680> command<31 00 FA 01 D6 00 00>
[14:30:15][E][daikin_rotex_can:042]: CanSensor::handleValue() => Sensor<t_hs> hex<0xFE0C> uint16<65036> float<6503.600098> out of range[1.000000, 90.000000]
[14:30:15][I][sendGet:146]: millis: 95172730|Vorlauftemperatur Heizung (TVBH) can_id<0x680> command<31 00 FA C0 FE 00 00>
[14:30:15][I][handle :146]: millis: 95172753|Vorlauftemperatur Heizung (TVBH)<39.799999> can_id<0x180> data<D2 00 FA C0 FE 01 8E> changed<0>
[14:30:15][I][sendGet:146]: millis: 95172756|ES mode can_id<0x680> command<31 00 FA 01 76 00 00>
[14:30:15][I][handle :146]: millis: 95172776|ES mode<> can_id<0x180> data<D2 00 FA 01 76 00 01> changed<0>
[14:30:15][I][sendGet:146]: millis: 95172782|Status Kompressor can_id<0x680> command<A1 00 61 00 00 00 00>
[14:30:15][I][handle :146]: millis: 95172795|Status Kompressor<> can_id<0x500> data<D2 00 61 00 00 00 00> changed<0>
[14:30:16][I][sendGet:146]: millis: 95173205|Durchfluss can_id<0x680> command<31 00 FA 01 DA 00 00>
[14:30:16][I][handle :146]: millis: 95173227|Durchfluss<0.000000> can_id<0x180> data<D2 00 FA 01 DA 00 00> changed<0>
[14:30:17][I][sendGet:146]: millis: 95174185|Betriebsmodus can_id<0x680> command<31 00 FA 01 12 00 00>
[14:30:17][I][handle :146]: millis: 95174211|Betriebsmodus<Sommer> can_id<0x180> data<D2 00 FA 01 12 05 00> changed<0>
[14:30:17][I][sendGet:146]: millis: 95174216|Leistung WW can_id<0x680> command<31 00 FA 06 68 00 00>
[14:30:17][I][handle :146]: millis: 95174239|Leistung WW<1.000000> can_id<0x180> data<D2 00 FA 06 68 00 64> changed<0>
[14:30:20][I][sendGet:146]: millis: 95177649|Ruecklauftemperatur Heizung can_id<0x680> command<31 00 FA C1 00 00 00>
[14:30:20][I][handle :146]: millis: 95177671|Ruecklauftemperatur Heizung<36.700001> can_id<0x180> data<D2 00 FA C1 00 01 6F> changed<0>
[14:30:20][I][sendGet:146]: millis: 95177674|Heizkreis Vorlauf (TV) can_id<0x680> command<31 00 FA C0 FC 00 00>
[14:30:20][I][handle :146]: millis: 95177714|Heizkreis Vorlauf (TV)<38.900002> can_id<0x180> data<D2 00 FA C0 FC 01 85> changed<0>
[14:30:20][I][sendGet:146]: millis: 95177766|Vorlauftemperatur Heizung (TVBH) can_id<0x680> command<31 00 FA C0 FE 00 00>
[14:30:20][I][handle :146]: millis: 95177790|Vorlauftemperatur Heizung (TVBH)<39.799999> can_id<0x180> data<D2 00 FA C0 FE 01 8E> changed<0>
[14:30:22][I][sendGet:146]: millis: 95179221|Betriebsmodus can_id<0x680> command<31 00 FA 01 12 00 00>
[14:30:22][I][handle :146]: millis: 95179244|Betriebsmodus<Sommer> can_id<0x180> data<D2 00 FA 01 12 05 00> changed<0>
[14:30:23][I][sendGet:146]: millis: 95180594|1 x Warmwasser can_id<0x680> command<31 00 FA 01 44 00 00>
[14:30:23][I][handle :146]: millis: 95180617|1 x Warmwasser<Aus> can_id<0x180> data<D2 00 FA 01 44 00 00> changed<0>
[14:30:24][I][PID:146]: millis: 95181574|thermal_power: sp: 0.000000, cv: -0.000000, e: 0.000000, p: 0.000381, i: 0.000057, d: -0.000001, map: 0.200000, fp: 0.001905, mfp: 0.001905, mp: 0.200000, o: 0.000437, val: 0.000437
[14:30:25][I][PID:146]: millis: 95181919|temperature_spread: sp: 2.200001, cv: 2.200000, e: 0.000001, p: 0.006502, i: 0.010000, d: -0.000005, map: 0.200000, fp: 0.032508, mfp: 0.032508, mp: 0.200000, o: 0.016497, val: 2.216497
[14:30:25][I][sendGet:146]: millis: 95182676|Ruecklauftemperatur Heizung can_id<0x680> command<31 00 FA C1 00 00 00>
[14:30:25][I][handle :146]: millis: 95182714|Ruecklauftemperatur Heizung<36.700001> can_id<0x180> data<D2 00 FA C1 00 01 6F> changed<0>
[14:30:25][I][sendGet:146]: millis: 95182720|Heizkreis Vorlauf (TV) can_id<0x680> command<31 00 FA C0 FC 00 00>
[14:30:25][I][handle :146]: millis: 95182739|Heizkreis Vorlauf (TV)<38.900002> can_id<0x180> data<D2 00 FA C0 FC 01 85> changed<0>
[14:30:25][I][sendGet:146]: millis: 95182807|Vorlauftemperatur Heizung (TVBH) can_id<0x680> command<31 00 FA C0 FE 00 00>
[... cut ...]
[14:30:32][I][sendGet:146]: millis: 95188985|T-Frostschutz can_id<0x680> command<31 00 FA 0A 00 00 00>
[14:30:32][I][handle :146]: millis: 95189007|T-Frostschutz<Aus> can_id<0x180> data<D2 00 FA 0A 00 FF 60> changed<0>
[14:30:32][I][sendGet:146]: millis: 95189012|DurchflussBer can_id<0x680> command<31 00 FA 06 9C 00 00>
[14:30:32][I][handle :146]: millis: 95189034|DurchflussBer<2.100000> can_id<0x180> data<D2 00 FA 06 9C 00 15> changed<0>
[14:30:32][I][sendGet:146]: millis: 95189038|Durchfluss Soll can_id<0x680> command<31 00 FA 06 89 00 00>

```

wrfz, 2025-04-15:
Is it possible that you had a negative value?

TNTLarsn, 2025-04-15:
> Is it possible that you had a negative value?

Is it? I'm not a heatpump expert

wrfz, 2025-04-15:
> Is it? I'm not a heatpump expert - Seems like the "error" is gone now

Did you have the error over a longer period of time?

`T-WE` and `Heizkreis Vorlauf (TV)` are exaktly the same values. (I don't know what the difference is)
Did you have the same problem there?

wrfz, 2025-04-15:
`0xFE0C` is the value sent by the HPSU.
- Either it was a negative value (this can be ruled out 100% in the case of flow temperature, because otherwise your condenser would have frozen and been damaged)
- or the HPSU got confused and sent an error value

In both cases I see no error in the ESP project code

TNTLarsn, 2025-04-17:
The project code and the heatpump, all work fine. 
Never mind...
Issue closed

closed 2025-04-17 by TNTLarsn