Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #93: Can't Compile

Bl00d-B0b, 2025-04-22:
It's clearly a problem with the external component. Otherwise you would be getting that error everywhere.
You need to include the header core/hal.h that defines that function.

The problem
Hello,
When I try to compile "manual download" firmware with ESPHome plugin to Home Assistant, then I receive error message: "error: 'millis' was not declared in this scope".

Which version of ESPHome has the issue?
2025.4.0

What type of installation are you using?
Home Assistant Add-on

Which version of Home Assistant has the issue?
2025.04.0

What platform are you using?
ESP32

Board
esp32-s3-devkitc-1

Component causing the issue
n/a

YAML Config
esphome:
  name: Name
  friendly_name: Long Name
  platformio_options:
    build_unflags:
    - "-std=gnu++11"
    - "-fno-rtti"
    build_flags:
    - "-std=gnu++17"

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

external_components:      # Automatically loads the *esphome-daikin-rotex-can* component from github during compilation
  - source: github://Trunks1982/Daikin-Rotex-HPSU-CAN

canbus:                   # Required for communication with the Daikin or Rotex heat pump
  - platform: esp32_can
    id: can_bus
    can_id: 0x680
    rx_pin: GPIO5
    tx_pin: GPIO6
    bit_rate: 20kbps    # HPSU heat pumps require a mandatory baud rate of 20kbs!

daikin_rotex_can:
  language: en
  id: rotext_hpsu
  canbus_id: can_bus
  project_git_hash:
    name: Project Git Hash
  entities:
    tdhw1:
      name: Hot Water Temperature

<skipped common settings>
Anything in the logs that might be useful for us?
In file included from src/esphome/components/daikin_rotex_can/entity.cpp:1:
src/esphome/components/daikin_rotex_can/entity.h: In member function 'bool esphome::daikin_rotex_can::TEntity::isChangedInLastMilliseconds(uint32_t) const':
src/esphome/components/daikin_rotex_can/entity.h:128:16: error: 'millis' was not declared in this scope
         return millis() > (getLastValueChange() + milliseconds);
                ^~~~~~
src/esphome/components/daikin_rotex_can/entity.h:128:16: note: suggested alternative: 'killpg'
[... 215 lines cut ...]
                     millis());
                     ^~~~~~
src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:550:21: note: suggested alternative: 'killpg'
                     millis());
                     ^~~~~~
                     killpg
src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:565:21: error: 'millis' was not declared in this scope
                     millis());
                     ^~~~~~
src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:565:21: note: suggested alternative: 'killpg'
                     millis());
                     ^~~~~~
                     killpg
src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:596:129: error: 'millis' was not declared in this scope
                     m_spread_error_detection.is_good_case_detected(), m_spread_error_detection.get_error_detection_timestamp(), millis());
                                                                                                                                 ^~~~~~
src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp:596:129: note: suggested alternative: 'killpg'
                     m_spread_error_detection.is_good_case_detected(), m_spread_error_detection.get_error_detection_timestamp(), millis());
                                                                                                                                 ^~~~~~
                                                                                                                                 killpg
*** [.pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/entity_manager.cpp.o] Error 1
*** [.pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/sensors.cpp.o] Error 1
*** [.pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/buttons.cpp.o] Error 1
*** [.pioenvs/alcove-hpsu/src/esphome/components/daikin_rotex_can/daikin_rotex_can.cpp.o] Error 1
========================== [FAILED] Took 3.67 seconds ==========================

wrfz, 2025-04-29:
Fixed. Thank you for the hint.
Please flash `dev` branch.

wrfz, 2025-04-29:
Please flash the `millis` branch now.

Bl00d-B0b, 2025-04-29:
Now it's successful. I assume you can now add to main :)

wrfz, 2025-04-29:
Thanks for testing!

I've merged the `millis` branch into `dev`.
It will be merged into `main` with the next release.

Bl00d-B0b, 2025-07-20:
Yes

closed 2025-07-20 by wrfz