Trunks1982/Daikin-Rotex-HPSU-CAN-Seriell #99: temperature setpoint for cooling

tosomax, 2025-05-31:
Hi!

Would it be possible to add the setpoint temperature of the water for the underfloor cooling mode?

Summer is approaching in Italy, I have already started using the cooling and it would be very useful to be able to regulate this parameter :)

Many thanks 🙏

wrfz, 2025-05-31:
There is no explicit setpoint for cooling mode. There is only one setpoint. Or missunderstood I something?

tosomax, 2025-05-31:
Actually I have one setpoint for the temperature of the hot water running through the underfloor heating system, and another setpoint used in summer when the operating mode is switched to "cooling"

In case I use a climatic curve for cooling, I need to set the two parameters "T-Flow Cooling max" and "Min T-Flow Cooling". Otherwise, if climatic curve is deactivated, I can set the water temperature with "T-Flow Cooling".

If you find it easier I can try to find the equivalent names in german in the manual :)

I understand that in Germany this might not be a priority since I don't think you would use much the cooling mode, but for us italians it would be awesome to have it :)

I have already started pumping cool water all day long here ahahahaha

[image]

wrfz, 2025-06-01:
Added all eight settings to the `dev` branch.

wrfz, 2025-06-01:
### German
```
    # Konfiguration -> HZK Konfig -> Kühlen
    start_t_out_cooling:
      name: Start Kühlen A-Temp
    max_t_out_cooling:
      name: Max Kühlen A-Temp
    t_flow_cooling_start:
      name: VL-Soll Start Kühlen
    t_flow_cooling_max:
      name: VL-Soll Max Kühlen
    min_t_flow_cooling:
      name: Min VL-Soll Kühlen
    t_flow_cooling:
      name: T-VL Kühlen
    t_h_c_switch:
      name: T_HK Umschaltung
    cooling_setpoint_adj:
      name: Kühlsollwert Korrektur
```

### English
```
    # Konfiguration -> HZK Konfig -> Kühlen
    start_t_out_cooling:
      name: Start T-Out Cooling
    max_t_out_cooling:
      name: Max T-Out Cooling
    t_flow_cooling_start:
      name: T-Flow Cooling start
    t_flow_cooling_max:
      name: T-Flow Cooling max
    min_t_flow_cooling:
      name: Min T-Flow Cooling
    t_flow_cooling:
      name: T-Flow Cooling
    t_h_c_switch:
      name: T_H/C Switch
    cooling_setpoint_adj:
      name: Cooling Setpoint adjustment
```

### Itialian
```
    # Konfiguration -> HZK Konfig -> Kühlen
    start_t_out_cooling:
      name: Avvio raffrescamento T esterna
    max_t_out_cooling:
      name: Max T esterna per raffrescamento
    t_flow_cooling_start:
      name: Mandata target avvio raffrescamento
    t_flow_cooling_max:
      name: Mandata target max raffrescamento
    min_t_flow_cooling:
      name: Mandata target min raffrescamento
    t_flow_cooling:
      name: Temperatura mandata raffrescamento
    t_h_c_switch:
      name: Commutazione riscaldamento/raffrescamento
    cooling_setpoint_adj:
      name: Correzione setpoint raffrescamento
```

If something about the translations doesn’t seem right, please let me know so I can adjust it promptly.

wrfz, 2025-07-18:
@tosomax Can this issue be closed?